seemoo-lab/nexmon #2: How to make it work on Kali ?

l-s-t, 2016-11-25:
Hello,

Is it planned to make it work on kali/ raspberry pi 3 ?

Thx, 

LST.

aagallag, 2016-12-14:
I actually was able to hack Nexmon into the Kali build scripts.  Although this is actually building from the  seemoo-lab/bcm-rpi3 repo.  I will try and figure out a way to compile from this repo, but at the moment, the Readme only provides steps for compiling directly on the Raspberry Pi.

Source Branch: https://github.com/aagallag/kali-arm-build-scripts/tree/nexmon-rpi3
Precompiled Image: https://github.com/aagallag/kali-arm-build-scripts/releases

marktube, 2016-12-16:
if i want to compile the patch on my raspberry kali, i should get the kali kernel headers first and resolve the complex dependencies. Is it ?