seemoo-lab/nexmon #3: LD_PRELOAD problem

phatal, 2016-11-26:
All times when I try to use LD_PRELOAD I get 

> ERROR: ld.so: object 'libfakeioctl.so' from LD_PRELOAD cannor be preloaded (cannot open shared object file): ignored.

I get this error everywhere including when try to copy example.

Also I get error `nexutil: command not found, but I have fw_bcmdhd.bin in vendor folder `. I make all step by step as in instruction, but looks like I forget something? (Nexus5 Nethunter)

matthiasseemoo, 2016-11-26:
Did you install libfakeioctl.so to /system/lib and nexutil to /system/bin?

phatal, 2016-11-30:
I install them, through nexmon app from play market. Or I need to build them from source manually?

When I install them through app I can run nexutil command only like a `/system/bin/nexutil -m true` for example, but LD_PRELOAD throw error anyway even if I launch it through `LD_PRELOAD=/system/lib/libfakeioctl.so` if I trying run directly `nexutil -m true` i get error command not found

phatal, 2016-11-30:
and sometimes I getting `__nex_driver_io : error` usually after trying to run `LD_PRELOAD=/system/lib/libfakeioctl.so aireplay-ng --test wlan0`