seemoo-lab/nexmon #4: Play store all crashes upon start

AutinDroid, 2016-12-01:
The new nexmon app crashes the app upon start on my nexus 5 running CM14.1.

FabianKnapp, 2016-12-01:
I will look at it.

matthiasseemoo, 2016-12-01:
Thanks for your feedback. Officially, we only support Marshmallow ROMs,
however, we are going to start some tests with the current CM 14.1 nightly
to fix the problem.

AutinDroid, 2016-12-01:
Thank you. I would like to mention that the previous build worked on my device.

FabianKnapp, 2016-12-01:
The following error occurs on CM 14.1 nightly

`12-01 03:14:07.555 14618-14618/de.tu_darmstadt.seemoo.nexmon E/AndroidRuntime: FATAL EXCEPTION: main
                                                                               Process: de.tu_darmstadt.seemoo.nexmon, PID: 14618
                                                                               java.lang.UnsatisfiedLinkError: dlopen failed: library "libc++.so" not found
                                                                                   at java.lang.Runtime.loadLibrary0(Runtime.java:989)
                                                                                   at java.lang.System.loadLibrary(System.java:1530)
                                                                                   at de.tu_darmstadt.seemoo.nexmon.MyApplication.initLibs(MyApplication.java:362)
                                                                                   at de.tu_darmstadt.seemoo.nexmon.MyApplication.onCreate(MyApplication.java:329)
                                                                                   at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1025)
                                                                                   at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5379)
                                                                                   at android.app.ActivityThread.-wrap2(ActivityThread.java)
                                                                                   at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1529)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                   at android.os.Looper.loop(Looper.java:154)
                                                                                   at android.app.ActivityThread.main(ActivityThread.java:6095)
                                                                                   at java.lang.reflect.Method.invoke(Native Method)
                                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)`

I don't know a fast fix for now, maybe it is due to the fact, that it is a nightly version?

It is reasonable that nexmon v1.2 worked on your device since v2.0 has much more complexity including wireshark dissection engine and so on.
We will create a version of nexmon where you can install at least the firmware and the tools as in v1.2 even for devices running CM 14.1 nightly, e.g.

Would you take the Marshmallow route for now? We would like to get more feedback from guys like you :-)

AutinDroid, 2016-12-01:
Sure, would love to help with the nexmon project!

matthiasseemoo, 2016-12-06:
The new app version 2.0.1 from the Play Store works fine with Android 7.

closed 2016-12-06 by matthiasseemoo