seemoo-lab/nexmon #6: Can I use it for LG G3 cat6(LG-F460S which use bcm4339 chip)?

raretomato, 2016-12-06:
I followed the steps for Nexus 5(bcm4339 chip).
I used Nexmon app through the Google Play.
I also install the tools and firmware, however, since then, my phone has restarted after the rooting within few minutes(?) seconds(?). (I am using KingRoot for temporal rooting.)

Could you give me some advice for me? Should I install the permanent rooting program?

matthiasseemoo, 2016-12-06:
Did you check the firmware version of your original firmware before installing the new firmware? It should be equal to 6.37.34.43 or 6.37.34.40. You can also send us your original firmware file and we can take a look.

raretomato, 2016-12-06:
@matthiasseemoo Where can I find the original firmware? I did not check the firmware version.

DanielAW, 2016-12-06:
@raretomato 
So you have access via adb right? Do you use a stock ROM or Cyanogen Mod? The Wifi firmware is stored at `/vendor/firmware/fw_bcmdhd.bin` in the stock ROM of the Nexus 5.

RafaelRuske, 2017-01-20:
I also have a LG G3, albeit model D855. Suposedly it sports a bcm4339.
Fully perma-rooted and installed an AOSP Rom (closest to stock Nexus 5 you can get on the G3).

Browsing to /vendor/firmware/ there's only a few files with .mdt and .bXX extensions (where X are numbers) concerning someting called wcnss. Those files appear to be from the Qualcomm WCNSS signed to be used with TrustZone. Also found a WCNSS_qcom_wlan_nv.bin on another folder which appears to be loaded by the WCNSS as the firmware, but its only 28Kb in size, (a big difference from the 505Kb of the compiled monitor mode fw) which makes me think its not the correct one.

I have no idea how to proceed.

zilent1, 2017-01-21:
@RafaelRuske the rom you are using has an AOSP kernel which includes no LG customizations to the kernel. Lg stock kernel has the correct firmware.

y00ri, 2017-03-26:
Hi 
I have Lg G3 (D855 - bcm4339) with Android 6.0 (V30d) rooted.
Unfortunately when I start Nexmon I get the following information: “Sorry your device is not supported by Nexmon, as we require a Broadcom WiFi chip that is missing in your smartphone!” 
App Version: 2.1.3 
Tool installation status (marked in green): nexutil (2.1.3), rawproxy (2.1.3), rawproxyreverse (2.1.3)
Could you let me know how can I solve this issue?

j-v-bernstein, 2022-02-04:
@y00ri @raretomato 

Delete the app and make a mental notice, that it is unusable as long as it hasn't been updated (which looks very much as never).

Wipe your phones(make backups if needed), install and root your device fresh.Now build the patched firmware and all the necessary tools (+either everyone, or justthose you wanna have) with the known requirements. 

As help did I attach some posts

(bc most of the original instructions resp. tutorial  sucks²³):

(https://github.com/seemoo-lab/nexmon/issues/387#issuecomment-599077135)

https://github.com/seemoo-lab/nexmon/issues/305#issuecomment-509162193

https://github.com/seemoo-lab/nexmon/blob/master/README.md

The devices in the threads maybe slightly different, but except of the correct Broacom chipset in your phone,  doesn't it matter one bit in complete install(compile) 'from source'-tutorial. 

I hope this could be resolved.