seemoo-lab/nexmon #12: Nexus 5x support

duramato, 2017-01-02:
Can you please add support for the Nexus 5x ? 
The Nexus 6P is supported, I would expect it's little brother to also be.

DanielAW, 2017-01-02:
Sorry, the Wifi in the Nexus 5x consists of a Qualcomm chip, see also #5 This is most likely to be a totally different architecture and so on. Our framework does only work with broadcom chips for now.

closed 2017-01-02 by DanielAW

duramato, 2017-01-02:
Is there any plan for adding support for Qualcomm chips? I can't seem to find any tool like this one that works on the 5x (googling for 5x doesn't help much, I get Nexus 5 results)

DanielAW, 2017-01-02:
No. It took us about 1,5 years to come to the point where we are right now. Qualcomm chips would need to start from zero again.