seemoo-lab/nexmon #14: Use with tshark

andreas-ibm, 2017-01-04:
Hi,

I may still be doing something wrong, but it's less obvious now!

I'm trying to use the firmware with tshark (so I can use FIND-LF https://github.com/schollz/find-lf) but tshark says it doesn't support monitor mode.

I think it comes down to the following:

```16:51:05 socket(PF_PACKET, SOCK_RAW, 768) = 3
16:51:05 ioctl(3, SIOCGIFINDEX, {ifr_name="lo", ifr_index=1}) = 0
16:51:05 ioctl(3, SIOCGIWNAME, 0x7eb83054) = 0
16:51:05 ioctl(3, SIOCGIWPRIV, 0x7eb83034) = -1 EOPNOTSUPP (Operation not supported)
16:51:05 ioctl(3, SIOCGIWMODE, 0x7eb83034) = 0
16:51:05 ioctl(3, SIOCGIFFLAGS, {ifr_name="wlan0", ifr_flags=IFF_UP|IFF_BROADCAST|IFF_MULTICAST}) = 0
16:51:05 ioctl(3, SIOCSIFFLAGS, {ifr_name="wlan0", ifr_flags=IFF_BROADCAST|IFF_MULTICAST}) = 0
16:51:05 ioctl(3, SIOCSIWMODE, 0x7eb83034) = -1 EOPNOTSUPP (Operation not supported)
16:51:05 ioctl(3, SIOCSIFFLAGS, {ifr_name="wlan0", ifr_flags=IFF_UP|IFF_BROADCAST|IFF_MULTICAST}) = 0
16:51:05 close(3)                       = 0```

I'm wondering if this is related to you mentioning that it's not necessary to explicitly set it to monitor mode?

Wondering how easy it would be to allow the SIOCSIWMODE to get further...

Is there an easy way to test that it's "working"? tcpdump -i wlan0 gets me some data...

matthiasseemoo, 2017-01-04:
You should use LD_PRELOAD=libfakeioctl.so to hook the ioctl calls to SIOCSIWMODE and others. With the current version in this repository, you should also be able to set monitor mode from an application.

DanielAW, 2017-01-04:
@andreas-ibm 
You may want to look at the README how we used "tcpdump" there, using tshark *should* make no difference

andreas-ibm, 2017-01-05:
Thanks for your help Matthias!

dhdutil seems to be an Android thing? I grepped dmesg for a similar line but couldn't find any.  running nexutil -m2 succeeds, but nexutil -m still reports 0. (and more importantly, tcpdump still only lists EN10MB link type, no radio ones)

Any other ideas for checking it on a PI?

matthiasseemoo, 2017-01-05:
Oh, I though you were experimenting with a Smartphone. On the raspberry Pi you should currently always be in monitor mode, but there might be a problem with the current version of the firmware in the repo that Daniel is going to figure out.

DanielAW, 2017-01-05:
Using our driver you should have this file: `/proc/nexmon_consoledump`. You can `cat` this file to get the console of the firmware:
`cat /proc/nexmon_consoledump`

andreas-ibm, 2017-01-05:
ah-hah, no /proc/nex* files!

-rw-r--r-- 1 root root 371736 Jan  4 15:21 /lib/firmware/brcm/brcmfmac43430-sdio.bin

it's definitely the "new" bin file... 

the plot thickens, the kernel module wasn't installed by make install!

installed it, depmodded, modprobe -r and back again.

now I have a /proc/nexmon_consoledump
iwconfig wlan0 reports Monitor

tcpdump barfs:
```tcpdump -I -i wlan0 -L
tcpdump: wlan0: SIOCGIWPRIV with a zero-length buffer didn't fail!```

so it's kinda one-step-back at the moment!

Any further tips?

DanielAW, 2017-01-05:
Did you try to run `tcpdump? with libfakeioctl? If this is the case can you try again without it? You only need this library on smartphones where we did not modify the driver.

andreas-ibm, 2017-01-05:
can't see the

```
000000.183 nexmon (Jan 4 2017 10:45:55)
```

line... 
Here's the full dump

```
hndarm_armr addr: 0x18003000, cr4_idx: 0
000000.001
RTE (SDIO-CDC) 7.45.41.26 (r640327) on BCM43430 r1 @ 37.4/81.6/81.6MHz
000000.001 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.005 reclaim section 0: Returned 47716 bytes to the heap
000000.007 wlc_bmac_info_init: host_enab 1
000000.026 wl0: Broadcom BCM43430 802.11 Wireless Controller 7.45.41.26 (r640327)
000000.027 TCAM: 256 used: 179 exceed:0
000000.028 reclaim section 1: Returned 81268 bytes to the heap
000000.029 sdpcmd_dpc: Enable
000000.044 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
000000.136 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
000000.137 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
000002.045 wl0: wl_open
000002.075 wl0: wlc_enable_probe_req: state down, deferring setting of host flags
```

andreas-ibm, 2017-01-05:
yeah, tried with and without fakeioctl

DanielAW, 2017-01-05:
BTW: which image are you using on your RPI3? Raspbian, Kali Linux? Any of the images which we provided in another repo?

andreas-ibm, 2017-01-05:
annoyingly, whilst tcpdump complains that SIOCGIWPRIV didn't fail, tshark / dumpcap complain that SIOCGIWPRIV failed with EPERM.

I'm going to put the kernel module and firmware back to original for a little while and get back to the actual problem I was trying to solve (and use bluetooth instead of wifi).

DanielAW, 2017-01-05:
Ok, I just updated my test system to 4.4.38 and rebuild the firmware and driver. It sill works fine on my system. 
> I'm going to put the kernel module and firmware back to original for a little while and get back to the actual problem I was trying to solve (and use bluetooth instead of wifi).

It would still be interessting to know why it does not work on your system. If you whant we could still try to figure out whats wrong, but I thing I need more information to reproduce your problem:
- Do you get any usefull messages in the kernel log (dmesg)?
- Which version of tcpdump do you use? For me its:
```
tcpdump version 4.6.2
libpcap version 1.6.2
OpenSSL 1.0.1t  3 May 2016
```
- Which kernel modules are loaded? Apart from `brcmfmac` there should be also a `brcmutil` module loaded.

andreas-ibm, 2017-01-05:
OK, I'll try to get back to it today, thanks.

closed 2017-09-01 by DanielAW

Zzzec, 2020-07-14:
> Ok, I just updated my test system to 4.4.38 and rebuild the firmware and driver. It sill works fine on my system.
> 
> > I'm going to put the kernel module and firmware back to original for a little while and get back to the actual problem I was trying to solve (and use bluetooth instead of wifi).
> 
> It would still be interessting to know why it does not work on your system. If you want we could still try to figure out whats wrong, but I think I need more information to reproduce your problem:
> 
> * Do you get any usefull messages in the kernel log (dmesg)?
> * Which version of tcpdump do you use? For me its:
> 
> ```
> tcpdump version 4.6.2
> libpcap version 1.6.2
> OpenSSL 1.0.1t  3 May 2016
> ```
> 
> * Which kernel modules are loaded? Apart from `brcmfmac` there should be also a `brcmutil` module loaded.

> Using our driver you should have this file: `/proc/nexmon_consoledump`. You can `cat` this file to get the console of the firmware:
> `cat /proc/nexmon_consoledump`
> 
> **Update:**
> @andreas-ibm can you send us the output of the command above?

I use your driver on my pi zero w but there's still no /proc/nex* thing, I think the driver is working because I can use monitor mode. How can I fix this or is there another way to dump the console?

andreas-ibm, 2020-07-14:
Sorry, I can't help, I don't have this system anymore!