seemoo-lab/nexmon #15: bcm43438 nexmon firmware not working

nethunteros, 2017-01-05:
I'm unsure if this is just me having this issue but previous repo (bcm-rpi3) firmware works fine but firmware built from this repo doesn't work on my pi3.  It shows monitor mode in iwconfig and switches channels in airodump but it's not receiving any SSIDs.

For now I'm just using previous repo but I would like to know if this is an issue on my part or if anyone has tested bcm43438 built from this repo and had it work. (This is tested on kali linux only)

DanielAW, 2017-01-05:
@nethunteros 
We will look into that. Your image is based on Kali Linux for the Raspberry Pi (https://www.offensive-security.com/kali-linux-arm-images/) right?

nethunteros, 2017-01-05:
@DanielAW - Thanks for the quick feedback!  I'm running my own build script that fixes the symbolic link also (https://github.com/nethunteros/rpi3-kalimon/blob/master/pi3.sh#L653).  It does three things after:
1) Mount the built kali image as a chroot
2) Fakes the uname syscall to trick it into thinking its on arm/with correct kernel: https://github.com/nethunteros/rpi3-kalimon/blob/master/pi3.sh#L674
3) Builds nexmon firmware/module

I'm building the kernel headers also and this is on 4.4.38.  If the build is working for you on raspbian it must be something I'm doing wrong with kernel build (I think).

Here is the image I'm using: https://build.nethunter.com/rpi3-nexmon/kali-0.5-rpi2.img.xz

Right now my build still uses old firmware but I still build with this repo also.

DanielAW, 2017-01-06:
@nethunteros I tried building nexmon on your image. It says it runs kernel 4.4.**39** instead of 38. Building our driver with the 4.4.39 results in a .ko file which causes a kernel crash (visible over serial console).
Are you able to build the original driver from the sources based on 4.4.39? If this is the case we can just port our few differences on the `brcmfmac` driver based on 39.

DanielAW, 2017-01-06:
Ok the `brcmfmac.ko` which is located in the root directory of your image runs fine with the firmware build out of this repository.
Only building the driver seems to be a problem.

nethunteros, 2017-01-07:
The brcmfmac.ko is built using your previous repo (bcm-rpi3), with a slight modification. I copied your latest brcmfmac from this (nexmon) repo into bcm-rpi3 so the module should be the same.  

The only difference should be the firmware build (I think) which doesn't work for me.  But you said you built firmware and it worked for you?

closed 2017-01-07 by nethunteros