seemoo-lab/nexmon #18: Injection on 5 GHz channels does not work correctly with aireplay-ng

DanielAW, 2017-01-09:
Migrated from bcm-public

j3ll3, 2017-06-13:
Hi Daniel,

I am interested in the frame injection on 5ghz channels with aireplay-ng on a Nexus 5. Can you give me some more details on this feature so I can participate?
- How did you verify that the problem is only with aireplay-ng?
- Is it possible to use another tool to inject frames on 5ghz channels?

Thanks!

Best regards,
Jelle

matthiasseemoo, 2017-06-13:
Write a firmware patch that calls our sendframe function. you can for
example call it in a new ioctl that you can trigger with nexutil. then you
can verify that the firmware can inject frames. to verify that frames
coming from android are injected correctly, add some printf calls in the
injection.c file to debug it. you can use dhdutil consoledump to see the
output of the printfs.

j3ll3, 2017-06-16:
Hi Daniel,

Thanks for the quick start! I followed your advise and the sendframe function works perfect for 2.4Ghz :) On the other hand for 5ghz it seems that the sendframe function accepts the data but the frame is not transmitted. There is no error response in dhdutil and when running wireshark on the Nexus 5 the frames are visible but it seems that they are not (correctly) transmitted in the air.
Is it possible that the sendframe function and injection.c are correct but that there is an issue in setting the bmc4339 in monitor/injection mode for 5ghz?

Best regards,
Jelle

j3ll3, 2017-06-20:
Hi Daniel,

Just experimented with the code in the send_frame function. After some trial and error I found that when the data_rate is set to 15 in injection.c it works :)

I can explain why it works at data_rate = 15; but I have tested it on different 5ghz channels. Wireshark reports still a data rate of 6M equal to data_rate = 12.

I am trying to commit my code asap.

Best regards,
Jelle