seemoo-lab/nexmon #20: Kalimon: NexMon on Kali Linux

DanielAW, 2017-01-09:
@nethunteros build a Kali Linux image with NexMon build in, see: https://github.com/seemoo-lab/bcm-rpi3/issues/12#issuecomment-266228305

(Migrated from: bcm-public)

Jip-Hop, 2017-01-16:
Also check out the GitHub repo of the Kali Raspberry 3 Image with Nexmon built in here: https://github.com/nethunteros/rpi3-kalimon
The latest release https://github.com/nethunteros/rpi3-kalimon/releases/tag/0.6 is based on the nexmon repo instead of bcm-rpi3 repo.

I'll be trying it out later today.

Jarvis785, 2017-02-17:
Deauthing and jamming perticular wifi is working ... But wifijamner.py (automated jamming) is not working .. anyone please suggest which jamming script should i use ..which is comparable with raspberry