seemoo-lab/nexmon #29: Failure building on the RPI3

Jip-Hop, 2017-01-16:
I'm unable to build on the RPI3 following the instructions in the readme.
I've installed a fresh copy of Raspbian Lite using NOOBS.

Then I follow the instructions from nexmon as super user by first running `sudo su`.
The error arises during the `Compile a patched firmware: make` step.
```
BUILDING DRIVER brcmfmac/brcmfmac.ko (details: log/driver.log)
Make[1]: *** /lib/modules/4.4.34-v7+/build: No such file or directory. Stop.
Makefile:31: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
```

I've tried the same approach without running the instructions as root but that leads to the same error.

Alternatively I tried restoring the image from the old repo, https://github.com/seemoo-lab/bcm-rpi3/releases/download/0.3/nexmon_rpi_12-10-2016.tar.gz, but booting the Pi lead to kernel panic. Same with the Kali image from here: https://github.com/seemoo-lab/nexmon/issues/20. Also kernel panic when booting.

What am I doing wrong?

Jip-Hop, 2017-01-16:
Turns out there is no `4.4.34-v7+` folder in /lib/modules/ but there is a `4.4.38+` folder as well as a `4.4.38-v7+` folder. Haven't figured out how to get it working yet.

Jip-Hop, 2017-01-16:
I've been able to get the Kali image working by using a 16GB SD card. For some reason it didn't work when restoring it to an 8GB SD card. Still having problems when building nexmon on the RPI3 with Raspbian though.

Jip-Hop, 2017-01-16:
Turns out I had the same problem as https://github.com/seemoo-lab/nexmon/issues/13. The kernel was updated as part of the `apt-get update && apt-get upgrade`. Rebooting after this step loaded the updated kernel and from there I could finish the process of building nexmon.

I guess it would be a good idea to add the reboot as a step in the instructions, after the `apt-get update && apt-get upgrade` step. With a description like, "If the kernel has been updated by the previous step, reboot now. If you're unsure, reboot now."

I couldn't finish the last optional step though.

Optional: install nexutil: from the root directory of our repository switch to the nexutil folder: cd utilities/nexutil/. Compile and install nexutil: make && make install

```
gxx -o nexutil nexutil.c - DBUILD_ON_RPI -DVERSION=\"IT_VERSION|" -I../../patches/include
nexutil.c:39:25: fatal error: argp-extern.h: No such file or directory
 #include <argp-extern.h>
			 ^

compilation terminated.
Makefile:11: recipe for target 'libs/armeabi/nexutil' failed
make: *** [libs/armeabi/nexutil] Error 1
```

bakashinji, 2017-01-24:
Hey there,

I had the same problems... You can look at my fork, I fixed the compiler errors there and it should build fine with the standard raspbian.

OussamaRomdhane, 2017-01-24:
You need to reboot after executing `sudo apt-get update && sudo apt-get upgrade` then continue doing the other steps. That's how I do it every time.

bakashinji, 2017-01-25:
This problem, i.e., not finding argp-extern.h has nothing to do with rebooting after updating... o_O You might fix compiling failures due to kernel-headers after upgrades with that, but I don't see the relation

OussamaRomdhane, 2017-01-25:
```
BUILDING DRIVER brcmfmac/brcmfmac.ko (details: log/driver.log)
Make[1]: *** /lib/modules/4.4.34-v7+/build: No such file or directory. Stop.
Makefile:31: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
```

This problem is (the problem this issue was opened to solve)

bakashinji, 2017-01-25:
Alright, but he figured that out himself anyways ^^

OussamaRomdhane, 2017-01-25:
Oh I didn't read the whole issue, sorry. Thanks for telling me though :smile:

blacknose2010, 2017-02-13:
I do have a similar problem as Jip-Hop:
`  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.4.48-v7+/build: No such file or directory.  Stop.
Makefile:40: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2`

I did every step of the manual using root user and also did a reboot after apt-get update && etc... also did source setup again ....  am I missing something ?

Dengesiz07, 2017-02-17:
I am also having the same issue, any solution so far?

DanielAW, 2017-02-18:
@Jip-Hop which folders are there in `/lib/modules` instead? I think the script just uses the output of `uname -r` and appends that to `/lib/modules/`. 
@Jip-Hop @Dengesiz07 please write the output of `ls -la /lib/modules` and `uname -r` here.

Jip-Hop, 2017-02-18:
Sorry I can't check that anymore. I'm now using the Kali Linux build with Nexmon already pre-installed.

DanielAW, 2017-02-18:
Ok, I will close this issue for now.

closed 2017-02-18 by DanielAW

reedand, 2017-04-06:
This looks to still be an issue so adding this here in the hope it helps resolve it.

>  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.9.20-v7+/build: No such file or directory.  Stop.
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

`uname -r
4.9.20-v7+`

> ls -la /lib/modules
total 32
drwxr-xr-x  8 root root 4096 Apr  6 00:29 .
drwxr-xr-x 20 root root 4096 Mar 30 15:11 ..
drwxr-xr-x  3 root root 4096 Mar 19 19:15 4.4.50+
drwxr-xr-x  4 root root 4096 Apr  2 15:09 4.4.50-v7+
drwxr-xr-x  3 root root 4096 Mar 29 01:51 4.9.17+
drwxr-xr-x  3 root root 4096 Mar 29 01:51 4.9.17-v7+
drwxr-xr-x  3 root root 4096 Apr  6 00:30 4.9.20+
drwxr-xr-x  3 root root 4096 Apr  6 00:30 4.9.20-v7+

The issue appears to be that `/lib/modules/4.9.20-v7+/build` doesn't exist:

> ls -la /lib/modules/4.9.20-v7+
total 1840
drwxr-xr-x  3 root root   4096 Apr  6 00:30 .
drwxr-xr-x  8 root root   4096 Apr  6 00:29 ..
drwxr-xr-x 11 root root   4096 Apr  6 00:30 kernel
-rw-r--r--  1 root root 474682 Apr  6 00:30 modules.alias
-rw-r--r--  1 root root 488875 Apr  6 00:30 modules.alias.bin
-rw-r--r--  1 root root   4778 Apr  6 00:29 modules.builtin
-rw-r--r--  1 root root   6290 Apr  6 00:30 modules.builtin.bin
-rw-r--r--  1 root root 151117 Apr  6 00:30 modules.dep
-rw-r--r--  1 root root 219044 Apr  6 00:30 modules.dep.bin
-rw-r--r--  1 root root    302 Apr  6 00:30 modules.devname
-rw-r--r--  1 root root  60567 Apr  6 00:29 modules.order
-rw-r--r--  1 root root     55 Apr  6 00:30 modules.softdep
-rw-r--r--  1 root root 197088 Apr  6 00:30 modules.symbols
-rw-r--r--  1 root root 243290 Apr  6 00:30 modules.symbols.bin

...however it does in an earlier version:

> ls -la /lib/modules/4.4.50-v7+
total 1780
drwxr-xr-x  4 root root   4096 Apr  2 15:09 .
drwxr-xr-x  8 root root   4096 Apr  6 00:29 ..
lrwxrwxrwx  1 root root     33 Mar  3 14:38 build -> /usr/src/linux-headers-4.4.50-v7+
drwxr-xr-x  2 root root   4096 Apr  2 15:09 extra
drwxr-xr-x 11 root root   4096 Mar  3 15:26 kernel
-rw-r--r--  1 root root 453102 Apr  2 15:09 modules.alias
-rw-r--r--  1 root root 473033 Apr  2 15:09 modules.alias.bin
-rw-r--r--  1 root root   4574 Mar  2 16:52 modules.builtin
-rw-r--r--  1 root root   6025 Apr  2 15:09 modules.builtin.bin
-rw-r--r--  1 root root 150458 Apr  2 15:09 modules.dep
-rw-r--r--  1 root root 218283 Apr  2 15:09 modules.dep.bin
-rw-r--r--  1 root root    302 Apr  2 15:09 modules.devname
-rw-r--r--  1 root root  60444 Mar  2 16:52 modules.order
-rw-r--r--  1 root root     55 Apr  2 15:09 modules.softdep
-rw-r--r--  1 root root 184408 Apr  2 15:09 modules.symbols
-rw-r--r--  1 root root 228368 Apr  2 15:09 modules.symbols.bin

I'm guessing but it appears that perhaps the problem is there's no linux headers for the version reported by `uname -r` so the symobilic link 'build' pointing to the source doesn't exist either:

> ls -la /usr/src/
total 16
drwxr-xr-x  4 root root 4096 Mar 19 19:15 .
drwxr-xr-x 10 root root 4096 Mar 24 17:57 ..
drwxr-xr-x 24 root root 4096 Mar 19 19:15 linux-headers-4.4.50+
drwxr-xr-x 24 root root 4096 Mar 19 19:15 linux-headers-4.4.50-v7+

> apt-get upgrade raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... raspberrypi-kernel-headers is already the newest version.
Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

devsys, 2017-06-01:
reedand illustrated exactly what is happening at least for me and I suspect for others as well. it seems when Raspbian is updated, some of the newer kernels that get installed don't include the source, which then leads to these problems. I found that I can get the sources installed by installing and running rpi-source to download the files: https://github.com/notro/rpi-source/wiki

jouellnyc, 2019-03-31:
@devsys - that seemed to work for me as it got me past the errors.

Secretspaceprogram, 2019-09-24:
@devsys worked for me too! thanks