seemoo-lab/nexmon #35: Raspbian 8 build /bin/sh: 1: ccobjcopy: not found

hl-a-k, 2017-01-26:
I builded patches for bcm43438 on the RPI3 using Raspbian 8.
after:
       cd patches/bcm43438/7_45_41_26/nexmon/
       make

I got an error: Makefile:99: recipe for target 'brcmfmac43430-sdio.bin' failed
 and log/flashpatches.log

> make[1]: Entering directory '/home/pi/Downloads/nexmon/patches/bcm43438/7_45_41_26/nexmon'
> ccobjcopy -O binary -j .text.bl_flash_patch_179 gen/patch.elf gen/section.bin && dd if=gen/section.bin of=brcmfmac43430-sdio.bin bs=1 conv=notrunc seek=$((0x00001000 - 0x00000000))
> /bin/sh: 1: ccobjcopy: not found
> gen/flashpatches.mk:2: recipe for target 'brcmfmac43430-sdio.bin' failed
> make[1]: *** [brcmfmac43430-sdio.bin] Error 127
> make[1]: Leaving directory '/home/pi/Downloads/nexmon/patches/bcm43438/7_45_41_26/nexmon'

matthiasseemoo, 2017-01-26:
did you run "source setup_env.sh"?

hl-a-k, 2017-01-26:
Yes I did. And I have tried many time.
Still not work.

bakashinji, 2017-01-27:
Are you sure you did? What could also be the case: You used sudo, or sudo su after you already sourced. Using sudo doesn't work, and if you sudo su'd then you have to source again. That happened to me quite often for make install-firmware ...

hl-a-k, 2017-01-28:
Thanks bakashinji. I used sudo after sourced. 
Now , I "su" at beginning,  the problem is fixed.

closed 2017-01-29 by DanielAW