seemoo-lab/nexmon #36: RPI3: WLC always down

bakashinji, 2017-01-27:
Whenever I try to send a frame using nexmon on the rpi3 wlc is down

first of, when inserting the driver: 
`000000.056 wl0: wlc_enable_probe_req: state down, deferring setting of host flags`

secondly when calling sendframe, with your beacon example:
`000044.020 ERR: wlc down`

I am not sure if this has anything to do with the stuff I fixed to get the whole thing to compile, as it did not do so from the beginning, see #34 ... To try out if I broke something I used the current commit and started with your CT guide from there, but the result was the same

matthiasseemoo, 2017-01-27:
You need to make sure that the radio is on and the d11 core is running. you
can achieve this by making sure that you are in monitor mode or by
disabling minimum power consumption (mpc).

bakashinji, 2017-01-27:
`ifconfig wlan0 up` and it ran again, ty

closed 2017-01-27 by bakashinji