seemoo-lab/nexmon #41: Nexmon directory typo

creepersbane, 2017-02-12:
When attempting to compile/make, I get:

/buildtools/ucode_extractor/ucodeext -r bcmdhd_sta.bin -b 0x2D830 -l 0x8B82 -o ucode.bin
make[3]: /buildtools/ucode_extractor/ucodeext: Command not found

The build command should be:

./buildtools/ucode_extractor/ucodeext -r bcmdhd_sta.bin -b 0x2D830 -l 0x8B82 -o ucode.bin
^
(Period goes there)

matthiasseemoo, 2017-02-12:
i suppose you did not run "source setup_env.sh" before running make?

creepersbane, 2017-02-13:
I ran it 5 times.

bakashinji, 2017-02-13:
Did you use sudo to call make [won't work --> sudo su; source setup_env.sh etc]? And from where did you call make?

creepersbane, 2018-02-25:
This no longer relevant since I no longer need to use Nexmon (sorry!)

closed 2018-02-25 by creepersbane

guilleeh90, 2022-03-21:
Try change the root directory with this code in terminal:

"sudo grep -rl '$(Q)$(NEXMON_ROOT)' | xargs sed -i s@'$(Q)$(NEXMON_ROOT)'@/home/pi/nexmon@g"

where /home/pi is your user directory. It runs for me.

BenjaminUrquhart, 2022-03-23:
@guilleeh90 I'm not sure why you're commenting on a closed issue from 5 years ago but sure.