seemoo-lab/nexmon #42: Unable to build backup firmware for Nexus 5

phaelyn, 2017-02-12:
I've been attempting to build and install the firmware patch for a Nexus 5 for a few weeks now; unfortunately, I can't get past the backup step. When I run `make backup-firmware`, I get the following error: 
[image]
I'm using the newest version of the repository and am running Ubuntu 16.04 (in a VM) to a stock Nexus 5 on 6.0.1 with the bootloader unlocked. Could anyone point me towards where my issue may be?

Thanks in advance!

matthiasseemoo, 2017-02-12:
yes, read and understand the error message in your screenshot and you might
be able to solve your problem.

phaelyn, 2017-02-12:
Thanks for the quick response. I've read the error message a number of times, but I still don't know what to do about it; I've run the prior `make` steps mentioned in the guide, so I'm not sure why the original binary wouldn't exist already. I'm also not sure why `su` wouldn't exist. 

I just graduated from college and am using your repo to aid a research project -- while I'm no stranger to the tools your software enables, I'm very new to adb/putting custom firmware on Android devices. I appreciate your patience, I'm aware that I may be making some beginner mistakes but I simply wasn't able to fix them on my own.

matthiasseemoo, 2017-02-12:
did you root your device?

phaelyn, 2017-02-12:
I'm sorry -- I used your project back when you supplied a pre-compiled `boot.img` that didn't require root, so I thought that these patches would work the same way. Thank you for your time!

closed 2017-02-12 by phaelyn