seemoo-lab/nexmon #43: RPi3 /usr/bin/ld: cannot find -lnexio

erospalberti, 2017-02-16:
Hello, in this step

`Install nexutil: from the root directory of our repository switch to the nexutil folder: cd utilities/nexutil/. Compile and install nexutil: make && make install.`

i receive this error 

`root@raspberrypi:/home/pi/tools/nexmon/utilities/nexutil# make
gcc -static -o nexutil nexutil.c -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -DUSE_NETLINK -I. -I../../patches/include -I../libnexio -L../libnexio/ -lnexio -I../libargp
/usr/bin/ld: impossibile trovare -lnexio
collect2: error: ld returned 1 exit status
Makefile:15: set di istruzioni per l'obiettivo "nexutil" non riuscito
make: *** [nexutil] Errore 1
root@raspberrypi:/home/pi/tools/nexmon/utilities/nexutil#
`

what i do?

thk

matthiasseemoo, 2017-02-16:
build libnexio first.

erospalberti, 2017-02-16:
Ok, sorry I didn't see the libnexio folder in utilities.

Now it work

Thank you very much.

closed 2017-02-17 by DanielAW