seemoo-lab/nexmon #47: Raspberry Pi 3 Model B - going back to original firmware?

the-bobo, 2017-02-24:
In README.md you say:

> Note: It is not possible to connect to an access point anymore using our modified driver and firmware, if you whant to go back to the default behaviour you will need to load the original driver and firmware.

How do we load the original driver and firmware? Are the directions from https://github.com/seemoo-lab/bcm-rpi3 still accurate?

> Original firmware and driver
> + Optional: remove the monitor mode driver `rmmod brcmfmac`
> + `cp /root/brcmfmac43430-sdio.orig.bin /lib/firmware/brcm/brcmfmac43430-sdio.bin`
> + use `modprobe brcmfmac` to load the unmodified driver and firmware

I do not want to brick my Pi's wlan0 by mistake! Thank you!

matthiasseemoo, 2017-02-24:
We recently added the ability to use station mode with our firmware, but
forgot to update this part in the readme. if you use the newest firmware
you have to use nexutil to activate monitor mode.

You can simply create a manual backup of your wifi firmware.

the-bobo, 2017-02-24:
Thank you for fast reply!

> You can simply create a manual backup of your wifi firmware.

This means, before making any changes do something like:
`cp /lib/firmware/brcm/brcmfmac43430-sdio.bin ~/backups/brcmfmac43430-sdio.orig.bin` ?

And then to revert `cp` back and use `modprobe`?

Also: what kernel version does nexutil support? I am on `rpi-4.9.y` right now.