seemoo-lab/nexmon #49: Raspberry Pi Model 3 B - How to turn monitor mode off?

the-bobo, 2017-02-24:
I have successfully built nexmon and used `nexutil -m2` to get into monitor mode. How do I get out of monitor mode back into managed mode?

I tried `nexutil -m0` and then `sudo ifconfig wlan0 down` and `sudo ifconfig wlan0 up` but i still can't connect.

haensche, 2017-02-25:
How could i solve the problem on a raspberry?

```
root@raspberrypi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon/src# nexutil
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
```
Are there any missing files or permissions?

matthiasseemoo, 2017-02-25:
Did you compile and load the nexmon brcmfmac driver. you also need to run
nexutil as root

haensche, 2017-02-25:
Thank you!
Reloading the driver worked!

`root@raspberrypi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon# rmmod brcmfmac
root@raspberrypi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon# insmod brcmfmac/brcmfmac.ko
root@raspberrypi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon# nexutil
root@raspberrypi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon# nexutil -m2
`

jlcastillo, 2017-03-26:
were you able to change between managed and monitor modes without rebooting?

the only way I found required a reboot:

- FIX if wlan0 doesn't appear when you do 'ifconfig' or 'iwconfig'. We need to restore the original Wifi firmware and reboot:

	cp /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac43430-sdio.bin.orig /lib/firmware/brcm/brcmfmac43430-sdio.bin	
	reboot

- enter monitor mode

	source /home/pi/nexmon/setup_env.sh
	cd /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon
	make install-firmware
	nexutil -m2

- to kill any processes that may interfere with wlan management:

	airmon-ng check kill

- now we can do 

	- to monitor packets:
		
		tcpdump -i wlan0
	
	- to see statistics about monitored packets:
	
		airodump-ng wlan0

	- to get a list of all access points around:
		
		iwlist wlan0 scanning

spadejack014, 2019-07-02:
> 
> 
> I went from monitor to managed mode like this:
> 
> ```
> - go into managed mode from monitored mode (provided wlan0 is up and in monitor mode as shown by 'ifconfig' and 'iwconfig')
> 
> 	rmmod brcmfmac 
> 	cp /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac43430-sdio.bin.orig /lib/firmware/brcm/brcmfmac43430-sdio.bin	
> 	modprobe brcmfmac
> 	ifconfig wlan0 up
> ```
> 
> however I'm still unable to change wlan0 channel in managed mode using 'iwconfig'.

I followed steps without any changes.but I still cannot run it.
trying nexutil gives the same error.
root@raspberrypi:/home/pi# nexutil -m2
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
what should I do to deal with the problems?

spadejack014, 2019-07-03:
I run it on 3B ,4.19.50.
which bcm should i choose ? 43430 ,43438 , or 43455?

spadejack014, 2019-07-03:
@haensche 
how did you solve such problem?
I reload the driver but it didn't work .