seemoo-lab/nexmon #50: definitions.mk missing?

tomose, 2017-03-01:
Following instructions from here: https://github.com/seemoo-lab/nexmon

Raspbian 8 on RPI3 Model B.

pi@pi:~/nexmon/patches/bcm43438/7_45_41_26/nexmon $ sudo make
Makefile:4: /firmwares/bcm43438/7_45_41_26/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43438/7_45_41_26/definitions.mk'.  Stop.

Thoughts?

matthiasseemoo, 2017-03-01:
run source setup_env.sh

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

bakashinji, 2017-03-01:
also, use sudo su and then use source setup_env.sh

OussamaRomdhane, 2017-03-01:
Also you might find it helpful to reboot after the third step

closed 2017-03-09 by matthiasseemoo

Guisardo, 2021-06-16:
This is old but... I just run into a similar issue and the problem some missing links. To fix, it check the if the NEXMON_ROOT export from the setup_env is correctly set and change it if needed.
Hope it helps to someone!