seemoo-lab/nexmon #52: make nexutil get error

pinpang, 2017-03-04:
In my RPI3
`root@pi3Debian:~/nexmon/utilities/nexutil# uname -a
Linux pi3Debian 4.4.48-v7+ #964 SMP Mon Feb 13 16:57:51 GMT 2017 armv7l GNU/Linux`
when i make nexutil:
`root@pi3Debian:~/nexmon/utilities/nexutil# make
gcc -static -o nexutil nexutil.c -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -DUSE_NETLINK -I. -I../../patches/include -I../libnexio -
L../libnexio/ -lnexio -I../libargp
/usr/bin/ld: cannot find -lnexio
collect2: error: ld returned 1 exit status
Makefile:15: recipe for target 'nexutil' failed
make: *** [nexutil] Error 1`

Now, what can I do?

ukscone, 2017-03-04:
If I remember correctly from a few days ago I had to cd to the libnexio directory, do a make and then remake nexutil

matthiasseemoo, 2017-03-09:
should work now

closed 2017-03-09 by matthiasseemoo