seemoo-lab/nexmon #54: Rooting the device

Lyubna, 2017-03-08:
Hello,
I am working on Android emulator (Nexus5 API) and when I launch the app it asks me to root the device, I tried installing the app on the device itself too, it says "we need root to proceed", but you mentioned in the description that it doesn't need rooting, why am I getting this?

matthiasseemoo, 2017-03-08:
where exactly did we mention that nexmon does not need root?

Lyubna, 2017-03-08:
[image]
Well , the screenshot is took from the description.

Lyubna, 2017-03-08:
[image]
In the screenshot, it says you can do it without root privileges

closed 2017-03-08 by Lyubna

matthiasseemoo, 2017-03-08:
That is the special case to communicate with the firmware without root
priviledges. the intension is to allow apps to control the firmware without
having root priviledges, however, you first nees root to install the
firmware and then you need root to set a security cookie in the firmware,
otherwise every non-root app would have access to control the firmware.

btw. it does not make sense to try the firmware hack in the amdroid
emulator, as the wifi chip will not be emulated.