seemoo-lab/nexmon #58: rpi-kernel-headers does not create the build directory for make

knuxyl, 2017-03-12:
I'm reinstalling Raspbian at the moment but yesterday when I tried to build this I was getting an error saying that
/lib/something/something/4.4.7v+?/build was not found
I'm assuming this is installed with sudo apt-get install rpi-kernel-headers but it was installed, and it still did not make this directory after I reinstalled it. I've had this problem before on Ubuntu distros where the kernel headers package was not creating the build folder. I am not sure if this is a distro problem, lack of more instructions for building, or just an invalid build process. Any help would be appreciated.

knuxyl, 2017-03-12:
I ended up download the file
linux-headers-4.9.13-v7+_4.9.13-v7+-2_armhf.deb
from here
https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/

And manually extracted it and copying the /usr/ folder to /usr/ on the Pi and do

ln -s /usr/src/linux-headers-4.9.13-v7+/ /lib/modules/4.9.13-v7+/build

knuxyl, 2017-03-12:
I did all this and now I'm getting

ERROR: Kernel configuration is invalid.
	include/generated/autoconf.h or /include/config/auto.conf are missing.
	Run 'make oldconfig && make prepare' on kernel src to fix it.

When I run the make oldconfig in the kernel source I am getting way too many questions that are way over my head. Has anyone successfully compiled this on the latest Raspbian Jessie? (4.9.13-v7+ kernel)

The instructions posted do not work if you have done a sudo apt-get -y dist-upgrade.

closed 2017-03-12 by knuxyl

coffeetrainwork, 2018-01-21:
Hi knuxyl

I'm dealing with the same problem,
please tell me how to solve this problem

make[1]: *** /lib/modules/4.9.59-v7+/build: No such file or directory.
and
include/generated/autoconf.h or /include/config/auto.conf are missing.