seemoo-lab/nexmon #59: Too many errors on RPi build

knuxyl, 2017-03-12:
pi@raspberrypi:~/Desktop/nexmon $ source setup_env.sh 
pi@raspberrypi:~/Desktop/nexmon $ cd patches/bcm43438/7_45_41_26/nexmon/
pi@raspberrypi:~/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon $ make

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  CREATING DIRECTORIES obj, gen, log
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:136:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:154:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412), CHAN2G(2, 2417), CHAN2G(3, 2422), CHAN2G(4, 2427),
  ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[0].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[1].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[2].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[3].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[4].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
  CHAN5G(44), CHAN5G(46), CHAN5G(48), CHAN5G(52), CHAN5G(56),
  ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[5].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[6].band’)
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
[... 245 lines cut ...]
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6238:21: error: array subscript is not an integer
   kfree(wiphy->bands[IEEE80211_BAND_5GHZ]);
                     ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c: In function ‘brcmf_cfg80211_attach’:
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6315:18: error: array subscript is not an integer
  if (wiphy->bands[IEEE80211_BAND_2GHZ]) {
                  ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:22: error: array subscript is not an integer
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                      ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:43: error: ‘struct ieee80211_channel’ has no member named ‘ht_cap’
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                           ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:51: error: request for member ‘cap’ in something not a structure or union
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                                   ^
/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:7: warning: assignment from incompatible pointer type
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
       ^
make[2]: *** [/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [_module_/home/pi/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
pi@raspberrypi:~/Desktop/nexmon/patches/bcm43438/7_45_41_26/nexmon $

dotysan, 2017-03-20:
Did you install the build dependencies? `sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf`

luigixp, 2017-03-20:
I get this same exact error. I installed all the dependencies.
My kernel is: 
root@pithree:~/nexmon-master/patches/bcm43438/7_45_41_26/nexmon# uname -r
4.9.14-v7+
Think it might have something to do with the newer kernel?

dotysan, 2017-03-20:
Yep. You appear to be on `next` from `rpi-update` instead of `stable` from Raspbian 8 apt which is at 4.4.50-v7+ (which is working for me here--without `rpi-update`).

In 4.9/`next` it looks like they removed the ieee80211_band enum. https://github.com/raspberrypi/linux/commit/57fbcce37be7c1d2622b56587c10ade00e96afa3#diff-6788d961d43074bcd7966d231861f6b8

Until then, use 'stable'/4.4 and rebuild. Or send up a patch!