seemoo-lab/nexmon #60: Reaver and wifite not working properly

lupeeumeeu, 2017-03-12:
So i've been testing those tools on my ao but it seems that they don't work at all, it keeps prompting me that it failed to associate or request timeout and wifite the same and it stuck on 0/0 and does not show any progress, and deauth packets can't deauthenticate, I could do it only once, please help me because I couldn't find any solution for this, I'm using a rpi3.thank you

matthiasseemoo, 2017-03-12:
Then debug the problem and find out what causes it. If you want to use
these tools you should at least understand how they work.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

hy-l, 2017-03-14:
I just got the same problem.It seems that the   "frame injection" is the reason.
I tested with aireplay-ng on my rpi3 :
`root@raspberrypi:/home/pi# aireplay-ng -9 wlan0`
`01:55:39  Trying broadcast probe requests...`
`01:55:41  No Answer...`
`01:55:41  Found 5 APs`
`...`
Any solution?
Thanks

lupeeumeeu, 2017-03-14:
i don't know if it have something about but airmon shows ????? in driver section, i'll try this command after to see if i get the same

lupeeumeeu, 2017-03-15:
Yes I got the same issue, maybe can be what you said, the frame injection, because he does not have any problem with find networks and use airmon, but even the deauth packets it gets some difficult, so it seems to be the frame, maybe?

Tapmemer, 2017-03-28:
@hy-i did you run 

sudo nexutil -m2? this will enable radiotap headers aircrack-ng tools need to function properly

hy-l, 2017-03-28:
Yes, I did. I tried many times with both -m1 and -m2, but no success. airodump-ng works perfect, but all "injection" related operations are not working. (aireplay-ng fakeauth, reaver, etc.)

lupeeumeeu, 2017-03-28:
yes, exactly the same thing for me, i tested with raspbian and kalimon, both with the same problem

closed 2017-05-27 by lupeeumeeu