seemoo-lab/nexmon #61: build failure on Raspberry Pi 3

dotysan, 2017-03-20:
The instructions fail to list a dependency, qpdf. This causes failure thusly:
```
  COMPRESSING UCODE gen/ucode.bin => gen/ucode_compressed.bin
/bin/bash: zlib-flate: command not found
Makefile:135: recipe for target 'gen/ucode_compressed.bin' failed
make: *** [gen/ucode_compressed.bin] Error 127
```

However, a 2nd bug also exists. Attempting to re-run make succeeds! This is due to zero-length (and thus useless) ucode_compressed.bin.

matthiasseemoo, 2017-03-20:
Everything you need to solve this issue is contained in the error message.

closed 2017-03-20 by matthiasseemoo

dotysan, 2017-03-20:
Sorry, more specific here: The primary bug is on https://github.com/seemoo-lab/nexmon/blame/master/README.md#L73 which is missing the `qpdf` build dependency.

dotysan, 2017-03-20:
The *secondary* bug is that it doesn't rebuild ucode_compressed.bin if it's clearly bogus/empty.