seemoo-lab/nexmon #62: Build error on Pi3

esserc, 2017-03-20:
When I try to make nexutil I get an unknown type error. 
Is an include missing somewhere?
This is whet I get:

pi@raspberrypi:~/git/nexmon/utilities/nexutil $ make

cd ../libnexio && make
make[1]: Entering directory '/home/pi/git/nexmon/utilities/libnexio'
gcc -c libnexio.c -o libnexio.o -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -I../../patches/include
ar rcs libnexio.a libnexio.o
make[1]: Leaving directory '/home/pi/git/nexmon/utilities/libnexio'
gcc -static -o nexutil nexutil.c bcmwifi_channels.c -DBUILD_ON_RPI -DVERSION=\"IT_VERSION\" -DUSE_NETLINK -I. -I../../patches/include -I../libnexio -L../libnexio/ -lnexio -I../libargp
In file included from ./typedefs.h:343:0,
                 from nexutil.c:61:
./bcmdefs.h:195:2: **error: unknown type name ‘uint’**
  uint origsize; /* Size of the virtual packet */
  ^
./bcmdefs.h:196:2: **error: unknown type name ‘uint’**
  uint nsegs;
  ^
In file included from nexutil.c:62:0:
./bcmwifi_channels.h:454:27: **error: unknown type name ‘uint’**
 extern int wf_mhz2channel(uint freq, uint start_factor);
                           ^
./bcmwifi_channels.h:454:38: **error: unknown type name ‘uint’**
 extern int wf_mhz2channel(uint freq, uint start_factor);
                                      ^
./bcmwifi_channels.h:482:27: **error: unknown type name ‘uint’**
 extern int wf_channel2mhz(uint channel, uint start_factor);
                           ^
./bcmwifi_channels.h:482:41: **error: unknown type name ‘uint’**
 extern int wf_channel2mhz(uint channel, uint start_factor);
                                         ^
./bcmwifi_channels.h:493:33: **error: unknown type name ‘uint’**
 extern uint16 wf_channel2chspec(uint ctl_ch, uint bw);
                                 ^
./bcmwifi_channels.h:493:46: **error: unknown type name ‘uint’**
 extern uint16 wf_channel2chspec(uint ctl_ch, uint bw);
                                              ^
./bcmwifi_channels.h:495:8: **error: unknown type name ‘uint’**
 extern uint wf_channel2freq(uint channel);
        ^
./bcmwifi_channels.h:495:29: **error: unknown type name ‘uint’**
 extern uint wf_channel2freq(uint channel);
                             ^
./bcmwifi_channels.h:496:8: **error: unknown type name ‘uint’**
 extern uint wf_freq2channel(uint freq);
        ^
./bcmwifi_channels.h:496:29: **error: unknown type name ‘uint’**
 extern uint wf_freq2channel(uint freq);
                             ^
Makefile:15: recipe for target 'nexutil' failed
make: *** [nexutil] Error 1
pi@raspberrypi:~/git/nexmon/utilities/nexutil $

matthiasseemoo, 2017-03-20:
just use this commit for now: https://github.com/seemoo-lab/nexmon/commit/87c10a7b5578ebb2cacd2e01d9b280a8338fd7f9

esserc, 2017-03-20:
No luck, I get the same errors. 
I thought uint was in /usr/include/arm-linux-gnueabihf/sys/types.h, which I believe is in the includes search path. What's missing?

Thanks

lemassykoi, 2017-03-20:
Hi all,

I got exactly the same problem, Raspberry Pi 3, Raspbian Lite fresh install, 4.4.50-v7+

nethunteros, 2017-03-20:
I added:
```
#include <sys/types.h>
#include <stdint.h>
#include <stdlib.h>
```
To top of nexutil.c and it built fine

lemassykoi, 2017-03-20:
I confirm. Thanks !

esserc, 2017-03-20:
I did that exact same thing, but it's still not working for me!

StephenWetzel, 2017-03-22:
I'm using the latest build (bd3791c), and nethunteros suggestion worked for me.  I added those includes above the define in nexutil.c (it didn't work when I added them below the define).

dmzimmerman, 2017-03-22:
What I've found so far is that if I manually load the kernel module and then use nexutil, nexutil works; but after a reboot, I see the same socket, bind and connect errors as lemassykoi until I manually unload and reload the kernel module - at which point nexutil works again. Something odd seems to be happening at boot time.

lemassykoi, 2017-03-23:
which commands do you enter to do unload/reload, please ?

dmzimmerman, 2017-03-24:
`rmmod brcmfmac` followed by `insmod brcmfmac/brcmfmac.ko` (while sitting in the `nexmon/patches/bcm43438/7_45_41_26/nexmon` directory).

lemassykoi, 2017-03-24:
Thanks. Here it is : 

`root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# rmmod brcmfmac
rmmod: ERROR: Module brcmfmac is not currently loaded
root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# insmod brcmfmac/brcmfmac.ko
insmod: ERROR: could not insert module brcmfmac/brcmfmac.ko: Unknown symbol in module`

I use this Pi for tests, maybe should I retry from scratch.

dmzimmerman, 2017-03-24:
It definitely seems like you're having a different issue than I am.

esserc, 2017-03-27:
I used the latest commit and added the includes. It worked! Thanks!

closed 2017-03-27 by esserc

ikbelkirasan, 2017-04-15:
I added 

`#include <types.h>`

to **nexutil.c** and it built fine.

amohan73, 2023-09-28:
since its C code, you can try using #ifndef statements to identify if a particular .h file is included or not and then #include these .. with all these libraries that you mentioned like stdlib or other, it's very common that some or the other file would include it..and double inclusion also will raise errors..
maybe that's why it works if you include at the top but not below already included files, as other files and libraries typically follow this protocol.

commit 03a428b3 (2017-05-23): fixes #77, fixes #62