seemoo-lab/nexmon #65: Nexus 5 app

mic-dot, 2017-03-22:
I'm running this app on my nexus 5, rooted 6.0.1  
I  have app version 2.1.3 installed and on the main screen ofthe app everything is green'd up saying i have nexutil,  rawproxy, and rawproxyreverse,  with nexmon firmware sucessfully installed.

I can can sucessfully use LD=PRELOAD from the terminal via my neuther install along with other wifi pen testing apps with no issues  

The issueis that I cannot use any of the tools in the nexmon app (wireshark, airodump, Airdecap, and aircrack) altought they show up in the sidebar. I did check mark and install them from the tools tab and can use them from the terminal and other apps as I mentioned. 

I'm guessing they are just not installed where the nexmon app is looking for them to be?

matthiasseemoo, 2017-03-22:
The nexmon app only needs nexutil, rawproxy and rawproxyreverse, all other
tools are reimplemented in java. maybe a restart of your phone helps?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

mic-dot, 2017-03-22:
Unfortunately no, I've rebooted it a handful of times since I installed it weeks ago. I'm thinking its an error on my end . Could someone tell me the steps I need to take to activate nonitor mode in the app

closed 2017-08-02 by DanielAW