seemoo-lab/nexmon #66: Raspberry Pi Model B - stuck at channel 1

jlcastillo, 2017-03-26:
I followed the instructions and successfully executed 'tcpdump -i wlan0' and 'airodump-ng wlan0'. However, I don't get data from channels other than 1.

- changing the channel manually with 'iwconfig wlan0 channel 11' returns

    Error for wireless request "Set Frequency" (8B04) :
    SET failed on device wlan0 ; Device or resource busy.

- I already removed all references to wlan0 in /etc/network/interfaces.
- I uninstalled wpa-supplicant ('apt-get remove wpasupplicant')
- I killed processes that may interfere ('airmon-ng check kill')

Did I miss anything?

haensche, 2017-03-26:
I've got the same issue.
I think that they haven't included channel switching yet. The error code is generated because wlan0 is managed via the nexmon patch and can no longer be accessed the normal way. Reboot and take the old driver module to configure wlan0 with iwconfig and iw.

Sources:
https://2016.mrmcd.net/fahrplan/system/event_attachments/attachments/000/002/897/original/nexmon_mrmcd16.pdf

Slide 29

jlcastillo, 2017-03-26:
Do you know how to do it without rebooting? So far I successfully changed from monitored to managed mode:

- go into managed mode from monitored mode (provided wlan0 is up and in monitor mode as shown by 'ifconfig' and 'iwconfig')

	rmmod brcmfmac 
	cp /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac43430-sdio.bin.orig /lib/firmware/brcm/brcmfmac43430-sdio.bin	
	modprobe brcmfmac
	ifconfig wlan0 up

... however I cannot change the channel with 'iwconfig wlan0 channel 10' or 'iw wlan0 set channel 10'.

Error for wireless request "Set Frequency" (8B04) :
    SET failed on device wlan0 ; Operation not supported.

JBaczuk, 2017-05-28:
I am also having a similar problem, though when I try the above commands to change channels I get:

```
Error for wireless request "Set Frequency" (8B04) :
    SET failed on device wlan0 ; Device or resource busy.
```

durika, 2017-06-07:
Got the same issue 
`
 CH 13 ][ Elapsed: 6 mins ][ 2017-06-07 15:56 ][ fixed channel wlan0: -1

 BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID

 BSSID              STATION            PWR   Rate    Lost    Frames  Probe

`

vzxrs, 2017-06-11:
changing type of monitor mode with nexutil solved this issue for me.
`nexutil -m0
nexutil -m1
nexutil -m2`
After this hoping seems work fine
Also try to switch through another modes (-m5, -m6, m7 even not existing modes) and then revert to -m2

jlcastillo, 2017-06-12:
vzxrs, were you able to use airmon to monitor all channels? Last time I tried I was able to receive some packets from adjacent channels, but not all. In the end I bought a small dongle that is working fine without hacks.

vzxrs, 2017-06-12:
jlcastillo, yes, by using this method I can use airodump-ng to monitor all channels with embedded rpi3 WiFi adapter

jlcastillo, 2017-06-13:
@vzxrs can you please write here all commands that you execute since the Raspberri Pi boots up until you get it to work with aurodump-ng ?  Can you also tell us the model of your Raspberri Pi?

vzxrs, 2017-06-14:
here it is

`sudo rmmod brcmfmac`
`sudo insmod <PATH TO KERNEL MODULE>/brcmfmac.ko` - kernel module is in /nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49 folder by default 
`nexutil -m7`
`sudo airodump-ng -i wlan0`

Here it outputs:
ioctl(SIOCSIWMODE) failed: Operation not supported
ioctl(SIOCSIWMODE) failed: Operation not supported
Error setting monitor mode on wlan0

Then

`nexutil -m2`
`sudo airodump-ng -i wlan0`

That's all

g0blinResearch, 2017-10-03:
Awesome, thanks @vzxrs - works a treat!

DH-Mike, 2017-10-20:
Little tip,
While in monitor mode (nexutil -m2),
You can manually change the channel on the interface by using the 'nexutil' binary like this:

nexutil -k$SOMECHANNEL

This works great on my RPI 3 B!

spadejack014, 2019-07-02:
which  bcm did you choose ? 
I failed with error "operation not supported"