seemoo-lab/nexmon #67: Can't get into monitor mode after reboot

StephenWetzel, 2017-03-29:
I have a Raspberry Pi 3, running Raspbian Jesse, and have gotten nexmon working, however when we reboot the Pi it stops working.  After rebooting running 'nexutil' gives this output:
```
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
```

ifconfig does not list the wlan0 interface, but iwconfig does:
```
pi@raspberrypi:~ $ iwconfig
wlan0     IEEE 802.11bgn  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

lo        no wireless extensions.

eth0      no wireless extensions.
```

ifconfig wlan0 up gives:
```
SIOCSIFFLAGS: Operation not permitted
```

dmesg | grep brcmfmac shows this:

```
[    3.774358] usbcore: registered new interface driver brcmfmac
[    3.906898] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    3.928639] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.014205] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    6.848886] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[    8.848925] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[    8.848948] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-52)
[   10.848915] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   10.848936] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[   13.598918] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   15.598917] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   17.598913] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   17.598937] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-52)
[   19.598927] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
```

We first tried this with NOOBS, but then reflashed with fresh Jessie Lite, and then installed nexmon, confirmed it worked, and then rebooted and got the exact same issue.

We saw this issue about reloading the firmware, but following those steps has not fixed the issue.
https://github.com/seemoo-lab/nexmon/issues/49#issuecomment-282483498
Running 
```
rmmod brcmfmac
modprobe brcmfmac  #insmod brcmfmac/brcmfmac.ko  has same result
```
Gives:
```
[  576.546295] WARNING: driver brcmfmac_sdio did not remove its interrupt handler!
[  577.222725] usbcore: deregistering interface driver brcmfmac
[  602.565904] brcmfmac: brcmf_sdiod_regrw_helper: failed to read data F1@0x08000, err: -84
[  602.570361] brcmfmac: brcmf_sdiod_regrw_helper: failed to read data F1@0x08000, err: -84
[  602.570379] brcmfmac: brcmf_chip_recognition: chip backplane type 15 is not supported
[  602.570391] brcmfmac: brcmf_sdio_probe_attach: brcmf_chip_attach failed!
[  602.570403] brcmfmac: brcmf_sdio_probe: brcmf_sdio_probe_attach failed
[  602.572444] brcmfmac: brcmf_ops_sdio_probe: F2 error, probe failed -19...
[  602.572695] usbcore: registered new interface driver brcmfmac
```

And then wlan0 no longer shows up at all in iwconfig:
```
lo        no wireless extensions.

eth0      no wireless extensions.
```

StephenWetzel, 2017-03-30:
So it turns out this was a manifestation of this issue:
https://github.com/seemoo-lab/nexmon/issues/61

When running make it failed complaining about zlib-flate, upon running it again it succeeded, but was using a 0 byte gen/ucode_compressed.bin file, which somehow worked the first time, but failed on reboot.  Installing qpdf, and then starting over from sources got it working again.  However, we still need to unload/reload the driver after each reboot, as well as include the header files in nexmon.c
https://github.com/seemoo-lab/nexmon/issues/62#issuecomment-287902835

For reference here is my current install script, with notes:
```
apt-get update && apt-get upgrade
apt install -y git libgmp3-dev gawk qpdf && apt install -y raspberrypi-kernel-headers
git clone https://github.com/seemoo-lab/nexmon.git
cd nexmon
source setup_env.sh   #Make sure you run this as root
make
cd patches/bcm43438/7_45_41_26/nexmon/
make #if this fails something about zlib-flate, you need qpdf, and make sure you delete the 0 byte gen/ucode_compressed.bin
make backup-firmware
make install-firmware
cd ../../../../
cd utilities/nexutil/
make #This fails complaining about uint, add includes (https://github.com/seemoo-lab/nexmon/issues/62#issuecomment-287902835)
make install
apt-get remove wpasupplicant
ifconfig
iwconfig
apt-get install aircrack-ng
nexutil -m2
airodump-ng wlan0 --berlin 10

#Reboot
shutdown -r now

#Reload driver
sudo rmmod brcmfmac
sudo insmod /home/pi/dev/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/brcmfmac.ko

nexutil -m2
airodump-ng wlan0 --berlin 10
```

hy-l, 2017-03-31:
Although I couldn't understand what you said very well, I just think this because the system auto load the original driver rather than the one you compiled after a reboot.
you need to find the original one and make a backup, then copy yours there:
`modinfo brcmfmac` #the first line should be the full path
`mv "<PATH TO THE DRIVER>/brcmfmac.ko" "<PATH TO THE DRIVER>/brcmfmac.ko.bak"`
`cp "<PATH TO YOU COMPILED ONE>/brcmfmac.ko" "<PATH TO THE DRIVER>/"`
and better do this:
`depmod -a`
then you can reboot

JBaczuk, 2017-04-08:
I have the same problem.  I don't understand what you mean @hy-l .  Isn't this supposed to be all done in the make backup-firmware and make install-firmware commands?  I found out that running those threw permission errors so I tried to copy them myself using sudo.

hy-l, 2017-04-09:
No, these commands only backup and install the firmware in '/lib/firmware/brcm/' directory, and `make install-firmware` also uses `rmmod` to unload the system's `brcmfmac.ko` kernel module and `insmod` to insert the one you compiled, but dose not replace the system one. So after a reboot, linux will use `modprobe` to auto load the original one in 
`/lib/modules/<KERNEL VERSION>/kernel/drivers/net/wireless/brcm80211/brcmfmac/`.

snfr067, 2021-07-07:
Excuse me, I tried "insmod /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_4.19.y-nexmon/brcmfmac.ko", but I get "insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format", does anybody can help?
My kernel is Linux raspberrypi 4.19.97-v7l+ #1293 SMP Wed Jan 22 17:16:14 GMT 2020 armv7l GNU/Linux

hy-l, 2021-07-07:
could it be because you stripped the module?

snfr067, 2021-07-08:
Thanks for your response.
It happened after I used command 'make install-firmware'.

> root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi# make install-firmware

         
         ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
     Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility

     COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
     COMPILING src/version.c => obj/version.o (details: log/compiler.log)
     PREPARING gen/nexmon.pre => gen/nexmon2.pre
     GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
     GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
     LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
     GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
     GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
     APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
     APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
     BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
     COPYING brcmfmac43455-sdio.bin => /lib/firmware/brcm/brcmfmac43455-sdio.bin
     RELOADING brcmfmac
     insmod: ERROR: could not insert module brcmfmac_4.19.y-nexmon/brcmfmac.ko: Invalid module format
     make: *** [Makefile:325: install-firmware] Error 1
`

Does 'stripped the module' meaning using the command 'sudo rmmod brcmfmac'?
I sure that I have never used this command and it hasn't been executed in 'make install-firmware'.
Here is the Makefile:

     install-firmware: brcmfmac43455-sdio.bin 
     ifeq ($(shell uname -m),$(filter $(shell uname -m), armv6l armv7l))
     	@printf "\033[0;31m  COPYING\033[0m brcmfmac43455-sdio.bin => /lib/firmware/brcm/brcmfmac43455-sdio.bin\n"
     	$(Q)sudo cp brcmfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
     ifeq ($(shell lsmod | grep "^brcmfmac" | wc -l), 1)
     	@printf "\033[0;31m  UNLOADING\033[0m brcmfmac\n"
     	$(Q)sudo rmmod brcmfmac
     endif
     	$(Q)sudo modprobe brcmutil
     	@printf "\033[0;31m  RELOADING\033[0m brcmfmac\n"
     ifeq ($(findstring 4.19,$(shell uname -r)),4.19)
     	$(Q)sudo insmod brcmfmac_4.19.y-nexmon/brcmfmac.ko
     else ifeq ($(findstring 5.4,$(shell uname -r)),5.4)
     	$(Q)sudo insmod brcmfmac_5.4.y-nexmon/brcmfmac.ko
     endif
     else
[... 44 lines cut ...]
     v4l2_common            16384  1 bcm2835_v4l2
     v4l2_mem2mem           24576  1 bcm2835_codec
     videobuf2_vmalloc      16384  1 bcm2835_v4l2
     bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
     videobuf2_dma_contig    20480  1 bcm2835_codec
     videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
     videobuf2_v4l2         24576  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
     videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
     videodev              200704  6 
     cm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
     media                  36864  3 bcm2835_codec,videodev,v4l2_mem2mem
     vc_sm_cma              36864  1 bcm2835_mmal_vchiq
     rpivid_mem             16384  0
     fixed                  16384  0
     uio_pdrv_genirq        16384  0
     uio                    20480  1 uio_pdrv_genirq
     i2c_dev                20480  0
     ip_tables              24576  0
     x_tables               32768  1 ip_tables
     ipv6                  454656  42
     

So I think I maybe didn't strip the module.
Is there any else to check if I stripped the module?
Thanks for your help!

snfr067, 2021-07-08:
By the way,
the interface 'wlan0' was disappered after using command 'make install-firmware'.
Is that means I stripped the module?
When I used command 'ifconfig wlan0 up', I got the result 'unknown interface wlan0'.
How can I get the interface 'wlan0' again?

hy-l, 2021-07-10:
no, 'strip the module' means that you ran the 'strip' command on the .ko file to reduce its size. i guessed so because it was the mistake i made before, but it doesn't seem to be the case here since you are unaware of it. perhaps it could be helpful if you can give the output of 'dmesg' after running insmod.

snfr067, 2021-07-12:
Thank you very much.
I have solved my problem.
I found that my kernel changed to 5.10 from 4.19 after I rebooted.
I think that it's because I have used command 'apt-get update' and 'apt-get install raspberrypi-kernel-headers'.
And I have gone on to install firmware without rebooting.
That's the reason why I get the result 'Linux raspberrypi 4.19.97' from 'uname'.
Afterwards I built a environment again without update kernel, and I install firmware successfully.
Thanks!