seemoo-lab/nexmon #69: MDK3 Not working on Nexmon RPi3 Image

leotakacs, 2017-04-02:
So I'm using the Nexmon Kali RPi3 Image from build.nethunter.net, and I installed MDK3. But when I do "mdk3 wlan0 d" (deauth) it says it's Disconnecting, but nothing happens. I ran "monstart", and iwconfig said that the wifi interface is in monitor mode so I don't know what's going on. I see mdk3 in the utilities folder of this project but for some reason I couldn't compile it.

leotakacs, 2017-04-05:
Also, everything works as expected on [this build]([url]https://github.com/nethunteros/rpi3-kalimon/releases/tag/0.6(url)) but I don't know for sure if it works on the Pi Zero W, as I am planning to purchase one soon for this purpose.

nethunteros, 2017-04-05:
The one in repo was built without nextuill/new drivers.  If you have a log to compare could you please add it?  Any other debugging would help to see what changed or is causing issue.