seemoo-lab/nexmon #70: Raspberry Pi Zero W promiscuous mode

dridri, 2017-04-04:
Hello, promiscuous does not seem to be working on pi0-W: tcpdump/libpcap only sees broadcasted frames (such as beacons), and injecting to arbitrary MAC address using libpcap does not emit anything.
I don't have a raspberry pi 3, so I can't say if it is a problem on pi0-W only.
The driver was built using NoobieDog's method ( #55 )

leotakacs, 2017-04-05:
Try this build. It's Kali Linux with the Nexmon driver already installed. It may work on the Pi Zero W, but I'm not sure as I have a Pi 3.

dridri, 2017-04-13:
Did you forget to add a link in your message or am I missing something ?
The actual problem is that I'm using a kernel compiled by myself so I cannot use a prebuilt system image

nethunteros, 2017-04-14:
I have a couple builds here:
https://build.nethunter.com/nexmon-pi-builds/

Not sure if it will make a difference in your case but it might help.

wortel538, 2017-05-22:
@nethunteros I need one for Raspian, if possible.. :(

thinktech2go, 2018-01-11:
https://linuxthor.github.io/projects/pi-zero-w-monitor-mode/

diegoroma0, 2020-11-17:
> 
> 
> I have a couple builds here:
> https://build.nethunter.com/nexmon-pi-builds/
> 
> Not sure if it will make a difference in your case but it might help.

HI! Thanks for share @nethunteros ! After trying many images this one works for me, but only on command line - terminal mode. I tried to start x but doesn't work. It have GUI installed? Am i doing something wrong?