seemoo-lab/nexmon #72: Compiling failed (libisl.so.10 doesn't exist)

M1z23R, 2017-04-14:
First of all hello everyone :D
I want to let you know that what you are doing is absolutely amazing, good job.
Unfortunately i am having problems with compiling on my Raspberry Pi 3 Model B under Parrot Os for Raspberry Pi.
I can compile buildtools without an error but when i try to build firmware i get this error
 COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
/media/usb/nexmon-master/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libisl.so.10: cannot open shared object file: No such file or directory
Makefile:48: recipe for target 'obj/ioctl.o' failed
make: *** [obj/ioctl.o] Error 1

I am trying to fix my gcc tools but i can't make it work, i am not that experienced at compiling thigs :/
If you can give me advice on what i can do i'd be grateful.

Also, as i said that i am inexperienced i want to know if there is a possibility to just download for example nexmon.driver or something like that and just install it without having to compile it ?

fsiep, 2017-04-26:
Same problem. I noticed that Nexcom brings cc1 and all the libraries with it. However libisl.so.10 is missing. Even installing it won't make a change since cc1 looks in a nexcom directory and the current version of the library is 13 (or higher) on the system I am using. Can that be fixed or a work-around provided? Thank you!

nethunteros, 2017-04-27:
@fsiep - Works for me:
```
ln -s /usr/lib/arm-linux-gnueabihf/libisl.so /usr/lib/arm-linux-gnueabihf/libisl.so.10
``

callen5914, 2017-06-04:
I don't quite understand that fix. Can you perhaps list out the commands including your step listed above.

chrise86, 2017-08-02:
Also facing this issue, and @nethunteros' symlink doesn't fix for me

Anty0, 2017-08-02:
Same problem on raspberry pi 3. I will try to fix it using fix above, but it isn't correct way, it should be fixed in repository...

chrise86, 2017-08-02:
Ended up using https://whitedome.com.au/re4son/re4son-kernel/

Anty0, 2017-08-02:
chrise86: Thank you for URL to re4son kernel. I got same problem as you described above. Re4son kernel is probably best solution now.

matthiasseemoo, 2017-08-21:
I added the sources for libisl version 0.10. Compile it as described in the README.md and the error is gone.

closed 2017-08-21 by matthiasseemoo

E3V3A, 2018-01-03:
@matthiasseemoo  
Why did you add the source for libisl v10 and not for 15, for example? 
Are you patching these sources, or can we use those that are in our kernel?

```bash
# ownerof /usr/lib/arm-linux-gnueabihf/libisl.so.15.3.0
libisl15:armhf: /usr/lib/arm-linux-gnueabihf/libisl.so.15.3.0

# apt show libisl15
Package: libisl15
Version: 0.18-1
Priority: optional
Section: libs
Source: isl
Maintainer: Debian GCC Maintainers <debian-gcc@lists.debian.org>
Installed-Size: 1,445 kB
Depends: libc6 (>= 2.4), libgmp10
Homepage: http://isl.gforge.inria.fr/
Download-Size: 426 kB
APT-Manual-Installed: no
APT-Sources: http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
Description: manipulating sets and relations of integer points bounded by linear constraints
 isl is a library for manipulating sets and relations of integer points
 bounded by linear constraints. Supported operations on sets include
 intersection, union, set difference, emptiness check, convex hull,
 (integer) affine hull, integer projection, and computing the lexicographic
 minimum using parametric integer programming. It also includes an ILP solver
 based on generalized basis reduction.
 .
 This package contains the runtime library.
```

matthiasseemoo, 2018-01-03:
The library does not seem to be downwards compatible. The arm compiler
version we have in our repository does not work with the newer version,
hence, i added the source code for the version that works.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

E3V3A, 2018-01-04:
@matthiasseemoo 
> The library does not seem to be downwards compatible. 

Do you mean the libisl15 (0.18.1) is not compatible with what? (Apart, obviously other kernels.)

> The arm compiler version we have in our repository does not work with the newer version,

Can you explain? 
1. Is it the problem with the compiler or with the dependencies?  OR
2. Is it a problem with how you implemented your patch code using ISL?

There seem to have been a lot of bug fixes and changes since libisl10 ( ISL version is ??).
If we can update/rewrite some of the code that uses ISL, we could solve this. Any idea where would be a good starting point to find all the isl dependent code?

matthiasseemoo, 2018-01-04:
Just try to use a newer libisl version while compiling patches and you will
see error messages. At least on recent versions of raspbian.

cy6or9, 2018-06-07:
same issue 
¨error while loading shared libraries: libisl.so.10: cannot open shared object file: No such file or directory
Makefile:59: recipe for target 'obj/ioctl.o' failed
make: *** [obj/ioctl.o] Error 1¨
any fix yet?

Oniat, 2021-02-23:
also facing the same issue here, but didnt have the same files as mentioned by @nethunteros 's instruction, but i did however had a different file by the same name,
```
┌──(root💀kali)-[/home/kali/Downloads/nexmon/patches/bcm4335b0/6.30.171.1_sta/nexmon]
└─# ls /usr/lib/arm-linux-gnueabihf/ | grep libisl
libisl.so.22
libisl.so.22.0.1
```
so I linked the libisl.so.22 with libisl.so.10 with 
```
# ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.22 /usr/lib/arm-linux-gnueabihf/libisl.so.10
```

But after doing so, i got the same error but with a different file
```
  COMPILING src/hndrte_debug.c => obj/hndrte_debug.o (details: log/compiler.log)
/home/kali/Downloads/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory
make: *** [Makefile:43: obj/hndrte_debug.o] Error 1
```
so i just did the same thing as before and looked for the file in /usr/lib/arm-linux-gnueabihf/ and linked the first file with it,
```
┌──(root💀kali)-[/home/kali/Downloads/nexmon/patches/bcm4335b0/6.30.171.1_sta/nexmon]
└─# ls /usr/lib/arm-linux-gnueabihf/ | grep libmpfr
libmpfr.so.6
libmpfr.so.6.1.0
```
```
ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4```
and voila!
![image](https://user-images.githubusercontent.com/37463800/108852852-4be84680-7610-11eb-878d-51c862356857.png)

commit 05d5d3bd (2017-08-21): Resolves #72