seemoo-lab/nexmon #75: Error with latest firmware

Y0Username, 2017-04-29:
**Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux**

/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:136:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:154:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412), CHAN2G(2, 2417), CHAN2G(3, 2422), CHAN2G(4, 2427),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[0].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[1].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[2].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[3].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[4].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
  CHAN5G(44), CHAN5G(46), CHAN5G(48), CHAN5G(52), CHAN5G(56),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[5].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[6].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[7].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[8].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[9].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
  CHAN5G(60), CHAN5G(64), CHAN5G(100), CHAN5G(104), CHAN5G(108),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[10].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[11].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[12].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[13].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[14].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
  CHAN5G(112), CHAN5G(116), CHAN5G(120), CHAN5G(124), CHAN5G(128),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[15].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[16].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[17].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[18].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[19].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
  CHAN5G(132), CHAN5G(136), CHAN5G(140), CHAN5G(144), CHAN5G(149),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for ‘__wl_5ghz_channels[20].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for ‘__wl_5ghz_channels[21].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for ‘__wl_5ghz_channels[22].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for ‘__wl_5ghz_channels[23].band’)
[... 221 lines cut ...]
   kfree(wiphy->bands[IEEE80211_BAND_5GHZ]->channels);
                                          ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6238:21: error: array subscript is not an integer
   kfree(wiphy->bands[IEEE80211_BAND_5GHZ]);
                     ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c: In function ‘brcmf_cfg80211_attach’:
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6315:18: error: array subscript is not an integer
  if (wiphy->bands[IEEE80211_BAND_2GHZ]) {
                  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:22: error: array subscript is not an integer
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                      ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:43: error: ‘struct ieee80211_channel’ has no member named ‘ht_cap’
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                           ^
[... cut ...]
                                                   ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:7: warning: assignment from incompatible pointer type
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
       ^
make[2]: *** [/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[1]: *** [_module_/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

cruzzer, 2017-05-01:
I'm seeing the same issue

Y0Username, 2017-05-01:
@cruzzer Do you have any other Wi-Fi adapter attached to your Pi?

cruzzer, 2017-05-01:
When I compile the software I did not.
Only after this post I've attached another adapter. Any outputs you would like me to collect?

Y0Username, 2017-05-01:
Which kernel version are you running?

neokore, 2017-05-02:
Same here. I've just downloaded Raspbian Lite from the Raspberry website and followed all instructions for Nexmon (including reboot).
The kernel installed after `apt update && apt upgrade` is 4.9.24-v7+ without rpi-installer.
rpi-issue (in case is useful): 
```
Raspberry Pi reference 2017-04-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, acc4f56597e75225b93be3e1e20504a40565a9b9, stage 2
```

cruzzer, 2017-05-02:
Version: 4.9.24-v7+
I went through the usual apt update, reboot, compile.

Y0Username, 2017-05-05:
I didn't get time to look at changes in the new firmware!
Don't upgrade the firmware.
I have downloaded the February build from the RPi archive: [raspbian-2017-02-27](http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-02-27/)
Follow the installation instructions in README. Instead using:  `apt update && apt upgrade` just use:  `apt update`

And it works on my Pi with the following kernel:
`Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux`

cruzzer, 2017-05-05:
The wording of "apt update" is admittedly a bit deceptive. It only updates the list of available software, not the software itself. In effect no software is updated, not just the firmware.

Y0Username, 2017-05-05:
@cruzzer Yes, you're right! `apt update` will populate the list of upgradable packages which is later used by `apt upgrade`.
What you're supposed to be doing is, run `apt update` and see the list using `apt list --upgradable` upgrade only necessary dependencies using `apt-get install <package_name>`.
However, the key here is to not to upgrade the `raspberrypi-kernel`

arrase, 2017-05-07:
Same bug here:

# uname -a
Linux raspberrypi 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux

morissette, 2017-05-07:
root@raspberrypi:/opt/nexmon/patches/bcm43438/7_45_41_26/nexmon# rpi-update 52241088c1da59a359110d39c1875cda56496764
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 12762  100 12762    0     0  52259      0 --:--:-- --:--:-- --:--:-- 52518
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.9.24-v7+

morissette, 2017-05-07:
Then after reboot

pi@raspberrypi:~ $ sudo su -
root@raspberrypi:~# nexpatch 
root@raspberrypi:/opt/nexmon/patches/bcm43438/7_45_41_26/nexmon# cd ../../../
root@raspberrypi:/opt/nexmon/patches# cd ..
root@raspberrypi:/opt/nexmon# source setup_env.sh 
root@raspberrypi:/opt/nexmon# nexpatch 
root@raspberrypi:/opt/nexmon/patches/bcm43438/7_45_41_26/nexmon# make

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  CREATING DIRECTORIES obj, gen, log
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.4.50-v7+/build: No such file or directory.  Stop.
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

morissette, 2017-05-07:
Then just delete raspberrypi-kernel-headers and reinstall