seemoo-lab/nexmon #81: could not insert module brcmfmac/brcmfmac.ko

ma1exe, 2017-05-11:
root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# make install-firmware

  CREATING DIRECTORIES obj, gen, log
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
  COPYING brcmfmac43430-sdio.bin => /lib/firmware/brcm/brcmfmac43430-sdio.bin
  RELOADING brcmfmac
insmod: ERROR: could not insert module brcmfmac/brcmfmac.ko: Invalid module format
Makefile:164: recipe for target 'install-firmware' failed
make: *** [install-firmware] Error 1
root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# uname -r
4.4.50-v7+

whats wrong?

BlueCocoa, 2017-05-14:
Same issue.

> root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# make
> 
> 
>           ###########   ###########   ##########    ##########           
>          ############  ############  ############  ############          
>          ##            ##            ##   ##   ##  ##        ##          
>          ##            ##            ##   ##   ##  ##        ##          
>          ###########   ####  ######  ##   ##   ##  ##    ######          
>           ###########  ####  #       ##   ##   ##  ##    #    #          
>                    ##  ##    ######  ##   ##   ##  ##    #    #          
>                    ##  ##    #       ##   ##   ##  ##    #    #          
>          ############  ##### ######  ##   ##   ##  ##### ######          
>          ###########    ###########  ##   ##   ##   ##########           
> 
>             S E C U R E   M O B I L E   N E T W O R K I N G               
> 
> 
>                                presents:                                  
> 
>               # ###   ###  #   # # ###  ###   ###  # ###                  
>               ##   # #   #  # #  ##   ##   # #   # ##   #                 
>               #    # #####   #   #    #    # #   # #    #                 
>               #    # #      # #  #    #    # #   # #    #                 
>               #    #  #### #   # #    #    #  ###  #    #                 
> 
>                 The C-based Firmware Patching Framework                   
> 
> 
>                            !!! WARNING !!!                                
>     Our software may damage your hardware and may void your hardware’s    
>      warranty! You use our tools at your own risk and responsibility      
> 
> 
>   CREATING DIRECTORIES obj, gen, log
>   COMPILING src/version.c => obj/version.o (details: log/compiler.log)
>   GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
>   GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
>   LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
>   GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
>   GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
>   APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
>   APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
>   BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
> root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# make backup-firmware
> cp /lib/firmware/brcm/brcmfmac43430-sdio.bin brcmfmac43430-sdio.bin.orig
> root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# make install-firmware
> 
> 
>           ###########   ###########   ##########    ##########           
>          ############  ############  ############  ############          
>          ##            ##            ##   ##   ##  ##        ##          
>          ##            ##            ##   ##   ##  ##        ##          
>          ###########   ####  ######  ##   ##   ##  ##    ######          
>           ###########  ####  #       ##   ##   ##  ##    #    #          
>                    ##  ##    ######  ##   ##   ##  ##    #    #          
>                    ##  ##    #       ##   ##   ##  ##    #    #          
>          ############  ##### ######  ##   ##   ##  ##### ######          
>          ###########    ###########  ##   ##   ##   ##########           
> 
>             S E C U R E   M O B I L E   N E T W O R K I N G               
> 
> 
>                                presents:                                  
> 
>               # ###   ###  #   # # ###  ###   ###  # ###                  
>               ##   # #   #  # #  ##   ##   # #   # ##   #                 
>               #    # #####   #   #    #    # #   # #    #                 
>               #    # #      # #  #    #    # #   # #    #                 
>               #    #  #### #   # #    #    #  ###  #    #                 
> 
>                 The C-based Firmware Patching Framework                   
> 
> 
>                            !!! WARNING !!!                                
>     Our software may damage your hardware and may void your hardware’s    
>      warranty! You use our tools at your own risk and responsibility      
> 
> 
>   CREATING DIRECTORIES obj, gen, log
>   COMPILING src/version.c => obj/version.o (details: log/compiler.log)
>   GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
>   GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
>   LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
>   GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
>   GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
>   APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
>   APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
>   BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
>   COPYING brcmfmac43430-sdio.bin => /lib/firmware/brcm/brcmfmac43430-sdio.bin
>   UNLOADING brcmfmac
>   RELOADING brcmfmac
> insmod: ERROR: could not insert module brcmfmac/brcmfmac.ko: Invalid module format
> Makefile:164: recipe for target 'install-firmware' failed
> make: *** [install-firmware] Error 1
> root@raspberrypi:/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon# uname -an
> Linux raspberrypi 4.4.48-v7+ #964 SMP Mon Feb 13 16:57:51 GMT 2017 armv7l GNU/Linux

vojtek, 2017-05-15:
same issue
Linux raspi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux

mferreira92, 2017-05-16:
Hi,

I'm having the same issue with kernel: Linux raspi 4.9.28+ 

Any workaround? I've testes with Raspbian (2017-04-10) but before the compila I've made the upgrade of all packages. Today will try with the default kernel of Raspbian, just to check if I'm able to compile.

vojtek, 2017-05-16:
solution - run
`sudo apt-get install git gawk qpdf flex bison`

Yanonix, 2017-05-16:
Same for me, with 4.4.50-v7+
@vojtek How do you perform your initial build on 4.9.28 without the issue #75 and the lastest commit ?
git, gawk, qpdf, flex, bison were already up-to-date (bison: 2:3.0.2.dfsg-2, flex: 2.5.39-8+deb8u2, qpdf: 5.1.2-2, gawk: 1:4.1.1+dfsg-1)

ma1exe, 2017-05-16:
@vojtek already done these commands, issue wasnt solved

lupeeumeeu, 2017-05-25:
same here, any solution? or a previous raspbian version??

lupeeumeeu, 2017-05-27:
i found the solution, for me at least, try following this, and don't use upgrade nor raspberrypi-kernel-headers:

sudo apt-get update

apt-get install dmks ncurses-dev libncurses5-dev flex bison git libgmp3-dev gawk qpdf
(i don't remeber wich one is the correct one ncurses or libncurses)

sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update

cd /usr/bin/rpi-source/

rpi-source

with this follow what is in the readme starting at git clone .....

lupeeumeeu, 2017-05-27:
but now i'm having some problems with channels, it stuck in -1, i can't change with iwconfig, i tried killing iplugd, but it keeps telling that it's busy, anyone knows how can i change?

vs4vijay, 2017-08-26:
Same issue... But can we get compiled kernel module(*.ko) file from somewhere? Why do we have to compile the kernel module ourselves?

matthiasseemoo, 2017-08-26:
As your kernel module always depends on your kernel version. We would have
to deliver modules for every possible kernel version. And if you want to
run a modified Wi-Fi firmware we  assume that you know how to compile a
kernel module.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

LostXine, 2017-08-27:
Same issue.
4.4.34-v7+

matthiasseemoo, 2017-08-28:
> @matthiasseemoo <https://github.com/matthiasseemoo> Thanks for the reply.
> Well I have already put a compiled kernel module for a version in my repo(
> https://github.com/vs4vijay/SwissArmyPi). I think it would be good for
> user if we put compiled modules for popular kernel versions. Because
> compiling requires us to download a whole lot of tools and whole nexmon
> repo(around 500 mb). I will manage a list of complied kernel modules. Thanks
>
> Repo. Link: https://github.com/vs4vijay/SwissArmyPi

In the end, it would be best if you would convince the maintainers of your
Linux distribution to include nexmon into their repositories. Then the
maintainers of the distribution could make sure to keep our modified driver
compatible to the currently delivered kernel.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

matthiasseemoo, 2018-09-23:
> @matthiasseemoo <https://github.com/matthiasseemoo> I think it would be
> @vs4vijay <https://github.com/vs4vijay> This is precisely how NOT to
> manage releases.
> In the ideal case nexmon would become a contrib package for one or more
> distributions, ensuring that we can get:
>    1. a source package relevant to the distribution+release we are using,
>    any time and forever,
>    2. distribution-specific tweaks and fixes which are discovered as
>    required for a given distribution+release, any time and forever,
>    3. version-specific binaries updated if necessary and as a given
>    distribution variant evolves (new kernel versions, headers/apis, etc.)
> This would all but end the current stream of people coming here again and
> again looking to do something they don't themselves have the skill, time,
> or desire to do, and would provide a growing community of professionals and
> enthusiasts with a stable release model (any time, and forever.)
> Since nexmon appears to be dependent on non-free firmware it's also likely
> this would be relegated to become "non-free contrib" -- and the unfortunate
> fact here is such packages tend to languish over time.
> Still, this would be better than "assuming" we can build kernel modules
> (and as an engineer who has been banging on linux for over 20 years,
> banging on code for over 30 years, I assure you it's not a question of IF I
> can, it's that *life is a limited resource* and if we're *all* doing this
> same B.S. work again and again -- then basically it amounts to an
> incredible loss of human effort, productivity, and lifetimes.
> "Yuck."
> [... 7 lines cut ...]
> Debian where once the initial investment is complete it will enable the
> community to continue and maintain that work over time (Without which it
> may never happen -- and it's possible project maintainers/contributors
> would never need to invest time in facilitating a contrib package ever
> again.)

Hi @wilson0x4d,

I like your idea of having a distribution package, however, it would need
to be updated for every new firmware update on each of the different
platforms. Additionally, we do not really have a team that is constantly
working on nexmon and I do not have the time to maintain a distribution
package.

It would already be helpful to get the nexmon patches for the brcmfmac
driver included into the official brcmfmac driver. This would relieve us
from having to patch this driver everytime we want to use it with a nexmon
firmware. However, our way of communicating with the driver might not be
the best. We inserted a netlink interface so that user space tools can talk
to the driver and exchange ioctl requests with the firmware. If there is a
better option for performing this task, we could also change our brcmfmac
patches and add new communication interfaces to the nexio library.

Our final goal would be to have a firmware that is also open source. The
old fullmac firmwares were quite similar to the brcmsmac functions for
softmac Wi-Fi cards. Hence, it would be nice to update the brcmsmac driver
to work with the new physical layer and MAC cores that are available in the
fullmac chips and then run this driver as firmware of the fullmac Wi-Fi
chips. Of course, we would need to also supply a run-time environment and
an SDIO/PCIE interface to the host. Having everything open, we could also
create open source distribution packages.

So, if any brcmfmac or brcmsmac developer or any other developer with the
proper experience is reading this thread and willing to help us build a
free firmware and driver environment for Broadcom chips, we are open to
discuss and work together on achieving this goal.

Matthias

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany