seemoo-lab/nexmon #82: driver installed on wlan1 instead of wlan0

JBaczuk, 2017-05-19:
I'm using a wifi usb device to get wifi connectivity which is on wlan1, but when I installed the driver, it appears to have installed on wlan1 instead of wlan0 (which is what I need) as evidenced by:

```
pi@raspberrypi:~ $ readlink /sys/class/net/wlan0/device/driver
../../../../../../../../bus/usb/drivers/rt2800usb
pi@raspberrypi:~ $ readlink /sys/class/net/wlan1/device/driver
../../../../../../../../bus/sdio/drivers/brcmfmac_sdio
```
How do I make sure the driver is on the correct interface (wlan0)?

JBaczuk, 2017-05-25:
Update: I've noticed it seems to be random which driver is loaded into wlan0 and wlan1 at startup.

JBaczuk, 2017-05-25:
Update: It actually loads the correct modules if the USB device is not connected at boot.

JuniorJPDJ, 2018-01-26:
You can just rename your interface with `ip link set <old name> name <new name>`
:D