seemoo-lab/nexmon #83: Pi W Zero Make Issue

drewablo, 2017-05-20:
When I run "Make" after cd into /nexmon/patches/bcm43438/7_45_41_26/nexmon, I get the following errors. I'm running Jessie Lite 4.9.24+. 

`  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:136:12: error: 'IEEE80211_BAND_2GHZ' undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:154:2: note: in expansion of macro 'CHAN2G'
  CHAN2G(1, 2412), CHAN2G(2, 2417), CHAN2G(3, 2422), CHAN2G(4, 2427),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: 'IEEE80211_BAND_5GHZ' undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro 'CHAN5G'
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for '__wl_5ghz_channels[0].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for '__wl_5ghz_channels[1].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for '__wl_5ghz_channels[2].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for '__wl_5ghz_channels[3].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for '__wl_5ghz_channels[4].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
  CHAN5G(44), CHAN5G(46), CHAN5G(48), CHAN5G(52), CHAN5G(56),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for '__wl_5ghz_channels[5].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for '__wl_5ghz_channels[6].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for '__wl_5ghz_channels[7].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for '__wl_5ghz_channels[8].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for '__wl_5ghz_channels[9].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
  CHAN5G(60), CHAN5G(64), CHAN5G(100), CHAN5G(104), CHAN5G(108),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for '__wl_5ghz_channels[10].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for '__wl_5ghz_channels[11].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for '__wl_5ghz_channels[12].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for '__wl_5ghz_channels[13].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for '__wl_5ghz_channels[14].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
  CHAN5G(112), CHAN5G(116), CHAN5G(120), CHAN5G(124), CHAN5G(128),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for '__wl_5ghz_channels[15].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for '__wl_5ghz_channels[16].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for '__wl_5ghz_channels[17].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for '__wl_5ghz_channels[18].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for '__wl_5ghz_channels[19].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
  CHAN5G(132), CHAN5G(136), CHAN5G(140), CHAN5G(144), CHAN5G(149),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for '__wl_5ghz_channels[20].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for '__wl_5ghz_channels[21].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: (near initialization for '__wl_5ghz_channels[22].band')
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:165:2: error: initializer element is not constant
[... 223 lines cut ...]
                                          ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6238:21: error: array subscript is not an integer
   kfree(wiphy->bands[IEEE80211_BAND_5GHZ]);
                     ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c: In function 'brcmf_cfg80211_attach':
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6315:18: error: array subscript is not an integer
  if (wiphy->bands[IEEE80211_BAND_2GHZ]) {
                  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:22: error: array subscript is not an integer
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                      ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:43: error: 'struct ieee80211_channel' has no member named 'ht_cap'
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                           ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:51: error: request for member 'cap' in something not a structure or union
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                                   ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:7: warning: assignment from incompatible pointer type
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
       ^
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[1]: *** [_module_/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
`

wortel538, 2017-05-21:
I'm having that too, even with using the version from february..

fossejc, 2017-05-28:
I have the same issue

fossejc, 2017-06-14:
I was able to compile on a RPI Zero W with kernel 4.9.24+
I used the [branch brcmfmac_kernel410](https://github.com/seemoo-lab/nexmon/tree/brcmfmac_kernel410) according to the instruction of issue #85 then I had to change `4.4.9-v7l+` by `4.9.24+` in file `patches/bcm43438/7_45_41_26/nexmon/Makefile`. Finally, I followed the steps described in #55 . The only difference is in file `utilities/libnexio/Makefile` line 5 instead of `ifeq ($(shell uname -m), $(filter $(shell uname -m), armv7l armv6l))` I had to put `ifneq ($(shell uname -m), $(filter $(shell uname -m), armv7l armv6l))` or it will try to compile for an android target.