seemoo-lab/nexmon #84: Connecting to an access point with the modified driver.

nikilp, 2017-05-22:
I want to use the modified driver to connect to an access point. The instructions say:
> **Note:** It is possible to connect to an access point using our modified driver and firmware, just set the wireless interface in Managed mode.
But when I set the wlan0 to managed mode (nexutil -m0), clicking the network interfaces icon in the top right gives me:
"No wireless interfaces found"
and the tooltip is:
"wlan0: Associated with
 wlan0 Configured 169.254.72.196/16"

Did anyone manage to connect with the modified driver to an access point?

lupeeumeeu, 2017-05-27:
did you remove wpasupplicant??don't do that if still want to use in managed mode, here i could switch easily

nikilp, 2017-06-04:
True. That was the issue. Thanks.

closed 2017-06-04 by nikilp