seemoo-lab/nexmon #85: Error with make on RPi3

stepj, 2017-06-04:
I am getting an error when trying to follow the guideline, especially while trying to make (at "xtract the ucode and flashpatches from the original firmware " step) 

below is the log 
====Quote============

root@raspberrypi:/home/pi/nexmon# source setup_env.sh 
root@raspberrypi:/home/pi/nexmon# make
  BUILDING BUILDTOOLS
make[1]: Entering directory '/home/pi/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o fpext.o -c fpext.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7.o -c darm/armv7.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7-tbl.o -c darm/armv7-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm.o -c darm/darm.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm-tbl.o -c darm/darm-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2.o -c darm/thumb2.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-decoder.o -c darm/thumb2-decoder.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-tbl.o -c darm/thumb2-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb.o -c darm/thumb.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb-tbl.o -c darm/thumb-tbl.c
gcc -o fpext fpext.o darm/armv7.o darm/armv7-tbl.o darm/darm.o darm/darm-tbl.o darm/thumb2.o darm/thumb2-decoder.o darm/thumb2-tbl.o darm/thumb.o darm/thumb-tbl.o
make[2]: Leaving directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/ucode_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o ucodeext.o -c ucodeext.c
gcc -o ucodeext ucodeext.o
make[2]: Leaving directory '/home/pi/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/home/pi/nexmon/buildtools/b43/assembler'
     DEPEND   dep/util.d
     YACC     parser.c
/bin/sh: 1: bison: not found
     DEPEND   dep/main.d
     DEPEND   dep/initvals.d
     DEPEND   dep/args.d
     YACC     parser.c
/bin/sh: 1: bison: not found
     CC       obj/args.o
     CC       obj/initvals.o
make[2]: *** No rule to make target 'parser.c', needed by 'dep/parser.d'.  Stop.
make[2]: Leaving directory '/home/pi/nexmon/buildtools/b43/assembler'
Makefile:4: recipe for target 'b43/assembler/b43-asm.bin' failed
make[1]: *** [b43/assembler/b43-asm.bin] Error 2
make[1]: Leaving directory '/home/pi/nexmon/buildtools'
Makefile:8: recipe for target 'buildtools' failed
make: *** [buildtools] Error 2
====un-Quote============

matthiasseemoo, 2017-06-04:
/bin/sh: 1: bison: not found

simply install bison.

BSDBlack, 2017-06-04:
With Raspbian on the RPi3 you need to install bison, as make also says:

>  /bin/sh: 1: bison: not found

stepj, 2017-06-04:
Thanks for the quick response. Now I am getting a different error
=======Q==========

root@raspberrypi:/home/pi/nexmon# make
  BUILDING BUILDTOOLS
make[1]: Entering directory '/home/pi/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
make[2]: 'fpext' is up to date.
make[2]: Leaving directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/ucode_extractor'
make[2]: 'ucodeext' is up to date.
make[2]: Leaving directory '/home/pi/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/home/pi/nexmon/buildtools/b43/assembler'
     YACC     parser.c
     LEX      scanner.c
/bin/sh: 1: flex: not found
     DEPEND   dep/parser.d
     DEPEND   dep/main.d
     LEX      scanner.c
/bin/sh: 1: flex: not found
     CC       obj/main.o
     CC       obj/parser.o
parser.c: In function ‘yyparse’:
parser.c:4330:18: warning: passing argument 1 of ‘yyerror’ discards ‘const’ qualifier from pointer target type
         yyerror (yymsgp);
                  ^
parser.y:26:13: note: expected ‘char *’ but argument is of type ‘const char *’
 extern void yyerror(char *);
             ^
     CC       obj/scanner.o
gcc: fatal error: no input files
compilation terminated.
Makefile:51: recipe for target 'obj/scanner.o' failed
make[2]: *** [obj/scanner.o] Error 4
make[2]: Leaving directory '/home/pi/nexmon/buildtools/b43/assembler'
Makefile:4: recipe for target 'b43/assembler/b43-asm.bin' failed
make[1]: *** [b43/assembler/b43-asm.bin] Error 2
make[1]: Leaving directory '/home/pi/nexmon/buildtools'
Makefile:8: recipe for target 'buildtools' failed
make: *** [buildtools] Error 2

BSDBlack, 2017-06-04:
now you need to install flex

> /bin/sh: 1: flex: not found

matthiasseemoo, 2017-06-04:
just read the error message and solve the problem like the last one.

stepj, 2017-06-04:
I am so sorry for these stupid questions, I am new to this. Now a new error while "Compiling a patched firmware"

===Q===
  CREATING DIRECTORIES obj, gen, log
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  COPYING UCODE /home/pi/nexmon/firmwares/bcm43438/7_45_41_26/ucode.bin => gen/ucode.bin
  COMPRESSING UCODE gen/ucode.bin => gen/ucode_compressed.bin
  GENERATING C FILE gen/ucode_compressed.bin => src/ucode_compressed.c
  COMPILING src/ucode_compressed.c => obj/ucode_compressed.o (details: log/compiler.log)
  GENERATING C FILE /home/pi/nexmon/firmwares/bcm43438/7_45_41_26/templateram.bin => src/templateram.c
  COMPILING src/templateram.c => obj/templateram.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/wrapper.c => obj/wrapper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/argprintf.c => obj/argprintf.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/helper.c => obj/helper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/udptunnel.c => obj/udptunnel.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/ucode_compression_code.c => obj/ucode_compression_code.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/objmem.c => obj/objmem.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/securitycookie.c => obj/securitycookie.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/radiotap.c => obj/radiotap.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/firmwares/bcm43438/7_45_41_26/flashpatches.c => obj/flashpatches.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  GENERATING LINKER FILE gen/memory.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.4.50-v7+/build: No such file or directory.  Stop.
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

stepj, 2017-06-04:
should I create a directory ??

BSDBlack, 2017-06-04:
No, you are missing some dependencies for compiling kernel modules:

As suggested in:
[https://stackoverflow.com/questions/20167411/how-to-compile-a-kernel-module-for-raspberry-pi#21043695](url)

you may install:
```
linux-image-rpi-rpfv
linux-headers-rpi-rpfv
```

using:
`sudo aptitude install linux-image-rpi-rpfv linux-headers-rpi-rpfv`

stepj, 2017-06-04:
Seems to be not an easy task since no free inodes are available at the RPI3 boot

===Logs===
pi@raspberrypi:~ $ sudo apt install initramfs-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
initramfs-tools is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up initramfs-tools (0.120+deb8u3) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.120+deb8u3) ...
ln: failed to create hard link ‘/boot/initrd.img-4.4.0-1-rpi.dpkg-bak’ => ‘/boot/initrd.img-4.4.0-1-rpi’: Operation not permitted

> cp: error writing ‘/boot/initrd.img-4.4.0-1-rpi.dpkg-bak’: No space left on device
> cp: failed to extend ‘/boot/initrd.img-4.4.0-1-rpi.dpkg-bak’: No space left on device

dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
====Logs===

===Logs2===
pi@raspberrypi:~ $ df -i
Filesystem      Inodes  IUsed   IFree IUse% Mounted on
/dev/root      1926848 169401 1757447    9% /
devtmpfs        117038    380  116658    1% /dev
tmpfs           118190     94  118096    1% /dev/shm
tmpfs           118190    484  117706    1% /run
tmpfs           118190      4  118186    1% /run/lock
tmpfs           118190     10  118180    1% /sys/fs/cgroup
/dev/mmcblk0p1       0      0       0     - /boot <<<
tmpfs           118190      6  118184    1% /run/user/1000
===End Logs2===

Thank you for the help. I will try to find a solution and update the thread if it works

BSDBlack, 2017-06-04:
I've just seen, you don't need all those libraries. In
[](https://github.com/seemoo-lab/nexmon/issues/75#issuecomment-299724461)

It seems to be enough to check out the headers an copy them there.

and a:
`apt-get clean`

removes all previous downloaded packages and frees some memory

stepj, 2017-06-04:
I did the clean and tried copying the headers. But still did not work! 

I am now trying to format the SD card and start from the beginning. I hope that works

stepj, 2017-06-04:
Ops! .. it seems that I overcome this error but there is another error as well in the same make
============================
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:136:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:154:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412), CHAN2G(2, 2417), CHAN2G(3, 2422), CHAN2G(4, 2427),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[0].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[1].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[2].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[3].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: error: (near initialization for ‘__wl_5ghz_channels[4].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
  CHAN5G(44), CHAN5G(46), CHAN5G(48), CHAN5G(52), CHAN5G(56),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[5].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[6].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[7].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[8].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:162:2: error: (near initialization for ‘__wl_5ghz_channels[9].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
  CHAN5G(60), CHAN5G(64), CHAN5G(100), CHAN5G(104), CHAN5G(108),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[10].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[11].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[12].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[13].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: initializer element is not constant
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:163:2: error: (near initialization for ‘__wl_5ghz_channels[14].band’)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: initializer element is not constant
  CHAN5G(112), CHAN5G(116), CHAN5G(120), CHAN5G(124), CHAN5G(128),
  ^
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:164:2: error: (near initialization for ‘__wl_5ghz_channels[15].band’)
[... cut ...]
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
       ^
make[2]: *** [/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [_module_/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
================
When I read 
> brcmfmac
I thought it is the same error. But the error now says 

> ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)

stepj, 2017-06-04:
According to this issue
https://github.com/seemoo-lab/nexmon/issues/59

Does that mean I have to downgrade my kernel or was it fixed ?

stepj, 2017-06-04:
OK, now following the commit in issue 59 I changed IEEE80211_BAND to NL80211_BAND and the errors seems to be reduced. 

Now I am left with this error 

```
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c: In function ‘brcmf_notify_escan_complete’:
/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:745:3: error: incompatible type for argument 2 of ‘cfg80211_scan_done’
   cfg80211_scan_done(scan_request, aborted);
   ^
In file included from /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:23:0:
./include/net/cfg80211.h:4304:6: note: expected ‘struct cfg80211_scan_info *’ but argument is of type ‘bool’
 void cfg80211_scan_done(struct cfg80211_scan_request *request,
      ^
make[2]: *** [/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [_module_/home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

```

Either I am getting closer to solve the issue or I am totally ruined

BSDBlack, 2017-06-07:
For the moment I am stuck with the same Problem, I hope we can find a solution

DanielAW, 2017-06-07:
You guys try to compile Nexmon on the the latest RPI3 kernel (4.9) right? Can you please try this branch: https://github.com/seemoo-lab/nexmon/tree/brcmfmac_kernel410  and report back if it works for you now? Don't get confused with the name of the branch, I just choose the wrong name :-/

BSDBlack, 2017-06-07:
Thank you very much for the branch.
I was able to compile the module and install it. The only flaw was, the Makefile puts the new module in `brcmfmac_kernel410/brcmfmac.ko` whereas `make install-firmware` awaits the module in `brcmfmac/brcmfmac.ko`

BSDBlack, 2017-06-07:
I've created a pull request which addresses the issue with the location
[https://github.com/seemoo-lab/nexmon/pull/90#issue-234250964](url)

DanielAW, 2017-06-07:
Thanks @BSDBlack I think I can merge this with master soon :-)

Obblet, 2017-06-09:
How can I clone a specific branch?

matthiasseemoo, 2017-06-09:
http://lmgtfy.com/?q=git+clone+branch

Obblet, 2017-06-09:
@DanielAW Yo dude I just wanted to thank you for this branch, this worked for me! 👍

stepj, 2017-06-12:
Thank you so much! .. That worked for me as well. 
Great Job!!

I think we may close this issue now after those confirmation

closed 2017-06-12 by stepj