seemoo-lab/nexmon #87: Raspberry Pi 3 'Make' Fails

LockedLogic, 2017-06-05:
I'm trying to use the instructions in the README.md to install the patch. Running `uname -a` gets a result of:

> Linux raspberrypi 4.9.30-v7+ #1001 SMP Fri May 26 16:09:18 BST 2017 armv7l GNU/Linux.
Running `/opt/vc/bin/vcgencmd version` gets:
> Copyright (c) 2012 Broadcom
version b8cdd5ae76f39d9f353dfa8fb48bf7e33b74903c (clean) (release)

I get as far as the first 'make'  (compile build tools and extract ucode and flashpoints) before an error occurs. Here's the terminal text:

> root@raspberrypi:/home/pi/Documents/nexmon# source setup_env.sh 
root@raspberrypi:/home/pi/Documents/nexmon# cd patches/bcm43438/
root@raspberrypi:/home/pi/Documents/nexmon/patches/bcm43438# cd 7_45_41_26/
root@raspberrypi:/home/pi/Documents/nexmon/patches/bcm43438/7_45_41_26# cd nexmon/
root@raspberrypi:/home/pi/Documents/nexmon/patches/bcm43438/7_45_41_26/nexmon# make

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware���s    
     warranty! You use our tools at your own risk and responsibility      

  CREATING DIRECTORIES obj, gen, log
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
/bin/bash: /home/pi/Documents/nexmon/buildtools/scripts/getdefine.sh: Permission denied
/bin/bash: /home/pi/Documents/nexmon/buildtools/scripts/getdefine.sh: Permission denied
/bin/bash: /home/pi/Documents/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-gcc: Permission denied
Makefile:48: recipe for target 'obj/ioctl.o' failed
make: *** [obj/ioctl.o] Error 126

Note that I have installed the necessary packages, set up the build enviroment and such. I installed a fresh version of Raspian very recently. which is why I suspect the make is failing. Can anyone tell me how to fix this?

DanielAW, 2017-06-06:
`/bin/arm-none-eabi-gcc: Permission denied`
It seems that you did not execute the commands as root, you should use `sudo su` as mentioned in the README.

I will close this for now because our version of the brcmfmac driver does not work wit the recent raspbian kernel `4.9.30-v7+`. We are currently working on this problem.

closed 2017-06-06 by DanielAW

LockedLogic, 2017-06-06:
It was run as root using sudo su. See Above:
>root@raspberrypi:/home/pi/Documents/nexmon# source setup_env.sh
>root@raspberrypi:/home/pi/Documents/nexmon/patches/bcm43438/7_45_41_26/nexmon# make

But, I suppose if the version isn't supported yet, it doesn't matter.