seemoo-lab/nexmon #88: Can't use WiFi in managed mode

0v3rfl00w, 2017-06-06:
Monitoring mode works fine. but can't use WiFi to connect even after monitor mode is turned off and the phone is rebooted : authentication error . I am on android 7.1.2 sprint  lg g3 (bcm4339)

0v3rfl00w, 2017-06-06:
is this file "/system/etc/firmware/fw_bcmdhd.bin" the only replaced file whene installing nexmon firmware ? If i need to restore the original wifi driver

matthiasseemoo, 2017-06-06:
yes, but sometimes using a patched firmware can result in a wired behaviour
that requires a complete reinstallation of the android rom (no need to
overwrite the data partition). however, with the nexmon firmware used in
the app, I did not observe this kind of error.

0v3rfl00w, 2017-06-06:
Is nexmon compatible with android 7 ? or should I downgrade to android 6 to fix this issue

matthiasseemoo, 2017-06-06:
As long as you have the correct driver version for the firmware version our
Nexmon firmware is based on, it should also work with Android 7. What is
the version number of your original Firmware version? You can for example
run "dhdutil consoledump" to figure this out.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

0v3rfl00w, 2017-06-07:
This is what i get :
start=0x001eb5d8 len=0x00000800
= 0
008181.972 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008182.015 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008182.057 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008182.100 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008182.142 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
lt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.163 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.205 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.248 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.290 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.333 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.375 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.419 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.461 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.504 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.546 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.589 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.631 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.674 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.716 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.760 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.802 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.845 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.887 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen = 0
008181.930 wlc_write_info_elt_safe, BUFLEN_CHECK_AND_RETURN: len = 14, maxlen

matthiasseemoo, 2017-06-07:
can you send me your original firmware file to mschulz@seemoo.de

0v3rfl00w, 2017-06-07:
Sorry i got confused . this was the command results with nexmon installed I will post the original one when I restore the system

0v3rfl00w, 2017-06-09:
@matthiasseemoo 
I sent the original file to your email 
I hope you could fix this issue

closed 2017-06-09 by 0v3rfl00w