seemoo-lab/nexmon #91: error compiling on 4.11.2 kernel

anemostec, 2017-06-10:
Hello, on a nano pi neo air with a 4.11.2 kernel, the compilation fails with the following error

root@NanoPi-NEO-Air:~/nexmon/patches/bcm43438/7_45_41_26/nexmon# make

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  CREATING DIRECTORIES obj, gen, log
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER  brcmfmac/brcmfmac.ko (details: log/driver.log)
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:136:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:154:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412), CHAN2G(2, 2417), CHAN2G(3, 2422), CHAN2G(4, 2427),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: initializer element is not constant
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: note: (near initialization for ‘__wl_5ghz_channels[0].band’)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:145:12: error: initializer element is not constant
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:161:14: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34), CHAN5G(36), CHAN5G(38), CHAN5G(40), CHAN5G(42),
[... 573 lines cut ...]
                     ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c: In function ‘brcmf_cfg80211_attach’:
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6315:18: error: array subscript is not an integer
  if (wiphy->bands[IEEE80211_BAND_2GHZ]) {
                  ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:22: error: array subscript is not an integer
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                      ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:43: error: ‘struct ieee80211_channel’ has no member named ‘ht_cap’
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                           ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:51: error: request for member ‘cap’ in something not a structure or union
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
                                                   ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.c:6316:7: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
   cap = &wiphy->bands[IEEE80211_BAND_2GHZ]->ht_cap.cap;
       ^
cc1: some warnings being treated as errors
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/chip.o] Error 2
make[1]: *** [_module_/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

DanielAW, 2017-06-11:
Hi,

you can try this branch (https://github.com/seemoo-lab/nexmon/tree/brcmfmac_kernel410) which is tested for kernel 4.9. In case it still dont work you need to apply some changes to the driver. Therefore copy the `cfg80211.c` and `cfg80211.h` from the original driver (https://github.com/raspberrypi/linux/tree/rpi-4.11.y) and use this patch (https://github.com/seemoo-lab/nexmon/blob/brcmfmac_kernel410/patches/bcm43438/7_45_41_26/nexmon/brcmfmac.patch) to apply our changes to the `cfg80211.c` file.

Let me know if this works for you.

Daniel

anemostec, 2017-06-13:
Hello and thanks, it is much better, but there is still one error after applying your patch by hand 

/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.c: In function ‘brcmf_cfg80211_add_iface’:
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.c:692:3: error: ‘infra’ undeclared (first use in this function)
   infra = 1;
   ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.c:692:3: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/chip.o] Error 2
make[1]: *** [_module_/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

DanielAW, 2017-06-13:
It seems something has gone wrong while applying the patch, the function `brcmf_cfg80211_add_iface` does not need the `infa` flag, see [here](https://github.com/seemoo-lab/nexmon/blob/brcmfmac_kernel410/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.c#L679).

anemostec, 2017-06-13:
right, I didn't apply the patch correctly, now here is the remaining error

scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/chip.o] Error 2
make[2]: *** Waiting for unfinished jobs....
scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/cfg80211.o] Error 2
make[1]: *** [_module_/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49] Error 2
Makefile:41: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

DanielAW, 2017-06-13:
This is very strange. I think this `scripts/basic/fixdep` is some script which belongs to the kernel itself. You did install the kernel headers, right?

anemostec, 2017-06-13:
oh, no I installed the headers from the kernel sources

marcosscriven, 2017-06-15:
@DanielAW - Just wanted to let you know I had the same compilation error, and [your branch](https://github.com/seemoo-lab/nexmon/tree/brcmfmac_kernel410) fixed it.

```
Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
```

anemostec, 2017-06-17:
@DanielAW Even with the correct header I still get compilation errors

/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c: In function ‘brcmf_sdio_dcmd_resp_wait’:
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c:1690:28: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  while (!(*condition) && (!signal_pending(current) && timeout))
                            ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c: In function ‘brcmf_sdio_bus_stop’:
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c:2414:3: error: implicit declaration of function ‘send_sig’ [-Werror=implicit-function-declaration]
   send_sig(SIGTERM, bus->watchdog_tsk, 1);
   ^
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c: In function ‘brcmf_sdio_watchdog_thread’:
/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.c:3947:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^
cc1: some warnings being treated as errors
make[2]: *** [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/sdio.o] Error 1
make[1]: *** [_module_/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49] Error 2
Makefile:42: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

DanielAW, 2017-06-17:
I'm not sure but it seems that `sched.h` moved from `include/linux/` to `include/linux/sched/`, so try moving the include to this new folder

anemostec, 2017-06-17:
mm no, those functions seem to be defined in linux/sched/signal.h for the 4.11 kernel
I added this file to the include definition of the file stdio.c and now I have linkages problem

  BUILDING DRIVER for kernel 4.9 brcmfmac/brcmfmac.ko (details: log/driver.log)
WARNING: "brcmu_d11_attach" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pkt_buf_get_skb" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_pdeq_tail" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_penq" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_mdeq" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_pdeq_match" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_peek_tail" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_flush" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_dotrev_str" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_penq_head" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_init" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pkt_buf_free_skb" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!
WARNING: "brcmu_pktq_mlen" [/root/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko] undefined!

anemostec, 2017-06-19:
I should have done something wrong as whenever I try to switch to monitor mode I get a segmentation fault
nexutil -m2
Segmentation fault

anemostec, 2017-06-19:
the error occurs on brcmf_fil_cmd_data_set
[  347.465528] 1ee0: 00000001 00000000 00000000 d4e51ed8 00000000 d4e50000 0000$
[  347.473783] 1f00: 00000000 c0609100 00087b50 c05c6c80 00000000 00000000 0000$
[  347.482039] 1f20: 00000000 00000011 c9ef6000 c05c841c 00000001 00000000 c086$
[  347.490295] 1f40: c9ef6000 00000000 00000008 c05cb544 de416e50 de102c38 0000$
[  347.498553] 1f60: c0b5632c 00000001 00000014 be94d674 00000008 dca1aa80 d4e5$
[  347.506807] 1f80: 00000008 00000004 be94d6c0 00000000 00000000 c05c6edc 0000$
[  347.515061] 1fa0: 00000000 c01074c0 be94d6c0 00000000 00000003 00087b70 0000$
[  347.523316] 1fc0: be94d6c0 00000000 00000000 00000121 00084d78 00000000 0000$
[  347.531571] 1fe0: 00000000 be94d604 00012881 000257a8 40070030 00000003 0000$
[  347.540099] [<bf1dc810>] (brcmf_fil_cmd_data_set [brcmfmac]) from [<c9ef7c00$
[  347.548527] Code: eaffffd8 bf1eea14 e92d40f0 e1a04000 (e5900000)
[  347.554994] ---[ end trace 0f922017d1ca8e80 ]---

DanielAW, 2017-06-23:
@anemostec 
I just fixed some bugs in the driver for kernel 4.9 (see 7fa41094a81b295b0be660bd81423f5e6dbb3818, c52cd02593b856e4d11d0242e4d30501cc6764e0 and 75348a1bf361ef7c2a7e202ce027500484fe66ce). You may apply these changes to your modified driver version.