seemoo-lab/nexmon #92: AP mode and monitor mode at once?

danz1, 2017-06-18:
First: thanks for the amazing work on this patch.

In order to use [hostapd-wpe](https://github.com/OpenSecurityResearch/hostapd-wpe), the wifi driver used needs to support creating two interfaces, one with monitor mode, and one with AP mode.

My system is: raspberry pi 3 b, [kali](https://whitedome.com.au/re4son/download/kali-linux-2017-1-armhf-re4son/), using hostapd-wpe from apt-get repositories (building my own makes no difference either), nexmon install as per the instructions in the image.

tcpdump works, as does nexutil and any other program using just the monitor mode. The errors start to occur when I try to use hostapd-wpe to creat an AP and monitor it after using nexutil -m2:
`nl80211: Set mode ifindex 6 iftype 3 (AP)`
`nl80211: Failed to set interface 6 to mode 3: -95 (Operation not supported)`
`nl80211: Try mode change after setting interface down`
`nl80211: Set mode ifindex 6 iftype 3 (AP)`
`nl80211: Failed to set interface 6 to mode 3: -95 (Operation not supported)`
`nl80211: Interface mode change to 3 from 0 failed`
`nl80211: Could not configure driver mode`
`nl80211: deinit ifname=wlan0 disabled_11b_rates=0`
`nl80211: Remove monitor interface: refcount=0`
`netlink: Operstate: ifindex=6 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)`
`nl80211: Set mode ifindex 6 iftype 2 (STATION)`
`nl80211: Failed to set interface 6 to mode 2: -95 (Operation not supported)`
`nl80211: Try mode change after setting interface down`
`nl80211: Set mode ifindex 6 iftype 2 (STATION)`
`nl80211: Failed to set interface 6 to mode 2: -95 (Operation not supported)`
`nl80211: Interface mode change to 2 from 0 failed`
`nl80211 driver initialization failed.`
`hostapd_interface_deinit_free(0x55df9b20)`
`hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1`
`hostapd_interface_deinit(0x55df9b20)`
`wlan0: interface state UNINITIALIZED->DISABLED`
`hostapd_bss_deinit: deinit bss wlan0`
`wlan0: AP-DISABLED`

Using nexmon in managed mode doesn't work either, as the monitor mode interface could not be created.

I was wondering whether you can get around this, in mode 2, or how easy it would be to support this.

henrikkorsgaard, 2017-08-01:
I second this request for the ability to create multiple interfaces, e.g. like how you would do it with iw: `sudo iw phy phy0 interface add mon0 type monitor`.

This would make the network interface with nexmon a bit more versatile IMO. I need monitor mode proximity detection (PCAP + RADIOTAP) and managed for exposing a small webserver/API on the RPI.

Thank you

matthiasseemoo, 2017-08-01:
That sounds mainly like a driver problem. Currently, when activating
monitor mode, we simple send the monitored frames in addition to frames
originating a connection as a station. Monitor frames are dropped in Linux'
Network stack as they do not contain valid data. To achieve your goal, you
would need to extend the driver by adding support for additional interfaces
and then exchange monitor mode frames only with the monitor interface and
send out all the other frames over the regular interface.

 Unfortunately, we do not have the time to develop this feature. However,
if you can do it, we would be happy to get your pull request ;-)

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

henrikkorsgaard, 2017-08-01:
Thank you for the quick reply.

Sadly, I do not have the necessary understanding of the involved steps or the experience with developing drivers to implement the feature. I fully understand why you do not have the time to do so either. 

Keep up the good work :)

brybalicious, 2018-03-13:
I have zero understanding of how/whether this is possible, but I would second the usefulness of being able to have 3 interfaces active on the Pi Zero - simultaneous wlan0 and ap0 (hostapd) as default, with the ability to put the device into monitor mode, and then revert back to simultaneous client/ap mode. This would enable you to e.g. penetration test a network, and then otherwise have the ability to reach your Pi to SSH into it, while also connecting to an AP on the other side of the interface, and forward packets to the AP.

At the moment, I have been able to set up simultaneous ap/client mode, and then (after installing nexmon firmware and driver) put the device into monitor mode, but not all at the same time. Perhaps the single physical interface can only support 2 out of the 3?