seemoo-lab/nexmon #93: Enabling Injection support on BCM4358 and BCM4356

matthiasseemoo, 2017-06-21:
We need your help to get injection working on the bcm4356 and the bcm4358. Transmitting frames from the firmware should work without any problem, but we currently cannot inject frames from Android. The problem is due to the PCI Express interface. On other chips such as the bcm4339, the chip is connected through SDIO to the smartphone. Here, the complete frames are copied from Android into the firmware's RAM and from there they are transmitted. With PCI Express, only the first 12 Bytes (length of the Ethernet header) are copied into the firmware's RAM, the rest stays in the RAM of the host system. To transmit the frame, a DMA copies a new WLAN header from the firmwares RAM together with the payload from the host RAM into the D11 core, from where the frame gets transmitted. That means, if we attach a Radiotap header longer than 12 Bytes for injection, we can only see the first 12 Bytes. To enable injection, we would need to understand how the PCI Express interface is configured and where we can increase the size from 12 Bytes to more. A first place to look would be the bcmdhd driver and there the pcie interface file. If you can figure out how it works, we could get injection running on these phones. Please, get in contact with us, if you could help us with this issue: Matthias Schulz <mschulz@seemoo.tu-darmstadt.de>

Gabrigawr, 2017-09-11:
Where would I download the file to start?

matthiasseemoo, 2017-09-11:
which file?

Gabrigawr, 2017-09-11:
The pcie file. I have a Nexus 6 that is my extra phone. Would I just go into the system file in there? I have Kali nethunter on it and Nexmon but realized it wouldn't work when I seen this.

matthiasseemoo, 2017-09-11:
do you have any reverse engineering experience? for the driver side there
is a pcie source file in the bcmdhd driver, but the part in the wifi
firmware needs reverse engineering.

matthiasseemoo, 2017-10-17:
The problem is now solved, I extended the libfakeioctl library by also hooking socket, bind, write and sendto functions and moved the extension into a new libnexmon library. Now, whenever a user injects raw frames either by calling write or sendto/send on a raw socket bound to the Wi-Fi interface, the frame is actually transferred using an ioctl to the Wi-Fi chip. By using this approach, the full frame ends up in the Wi-Fi chips RAM and can be fully processed similar to the SDIO case. I did not check performance especially when using high modulation rates and multiple spatial streams, but it seems to work quite reliable. aireplay-ng (in the end libosdep) calls the write function to send frames, libpcap uses the sendto or send functions to send frames.

closed 2017-10-17 by matthiasseemoo

flesh91, 2018-01-19:
Injection currently work on bcm4356 (nexus 6), or not? If  work, you need update information in readme.

matthiasseemoo, 2018-01-19:
You might need to port the ioctl handlers from bcm4358 to the other
firmware to be able to get injections working. Then you can preload
libnexmon.so to inject frames using ioctls.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

flesh91, 2018-01-20:
Thank you for your reply. I would be grateful if you tell me how to do this step by step

matthiasseemoo, 2018-01-20:
Then I would be faster doing it myself. Compare the code between the two
patches and copy the missing parts.

flesh91, 2018-01-20:
Honestly, I did not understand what I need to do. If you have the opportunity and the time, please do it yourself, or lay out a step-by-step manual, I'll be very grateful to you. Sorry for my bad english.

matthiasseemoo, 2018-01-20:
The lack of time is my current problem.

hussaini1, 2018-03-31:
Unable to install fw in nexus 6

matt42462, 2018-10-04:
Could ya, did ya, have ya? Maybe freed up some time to be awesome and port those ioctl handlers for the nexus 6 yet? And I also have a spr s6. I can jus use the firmware for the 6p yea?

superstuart, 2019-05-10:
Matthiasseemoo. I installed the Nexmon app on my 6P running Oreo 8.0.0. Monitor mode is dandy but whenever I try using packet injection it says 'No injection support available, sorry". I am reading the various online resources about the issue of getting packet injection working using the nexmon framework but I don't fully understand it yet. If you get the opportunity I would really appreciate your advice on how to go about doing it

matthiasseemoo, 2019-05-10:
Did you preload libnexmon.so

2019, 16:51:

DimitrisElGreco, 2022-12-25:
Hello dear developers,
any new updates?
is it implemented yet?