seemoo-lab/nexmon #95: Galaxy S4 (BCM4335) support

0xpr03, 2017-06-24:
Hi,
Galaxy S4 support for I [think BCM 4335](http://www.techinsights.com/about-techinsights/overview/blog/samsung-galaxy-s4-teardown/) would be really nice.
I would like to help, but I don't know exactly how. And as you have already support for higher bcm's I thought it should be "fairly" (TM) easy.

chrisdroid, 2017-06-25:
+1 BCM4335 chipset support. It is/was one of the most popular phones ever, +100 million units have been manufactured (much more than Nexus phones).

Latest firmware (6.30.171.1): [bcmdhd_sta.zip](https://github.com/seemoo-lab/nexmon/files/1100375/bcmdhd_sta.zip)

Firmware extracted from: https://download.lineageos.org/jfltexx

md5sum /system/etc/wifi/bcmdhd_sta.bin
3bd83a9b7679b67230d4b1c669be8a63  bcmdhd_sta.bin

4335b0-roml/sdio-ag-pool-p2p-pno-pktfilter-keepalive-aoe-sr-vsdb-proptxstatus-lpc-wl11u-betdls-autoabn-txbf-rcc-fmc-wepso-ve-ccx-okc-ltecx-clm_4335_ss-txpwr-noccxaka-mfp 
Version: 6.30.171.1 
CRC: 16cc5f59 
Date: Thu 2015-03-05 16:47:36 KST 
FWID 01-c016b565

dmesg on Samsung S4 with this firmware:

```
[   77.307556] Dongle Host Driver, version 1.141.64.13 (r)
[   77.307556] Compiled in drivers/net/wireless/bcmdhd
[   77.307556] wl_android_wifi_on in
[   77.307556] wifi_platform_set_power = 1
[   77.307586] check BCM4335, check_bcm4335_rev 
[   77.307647] brcm_wlan_power Enter: power on
[   77.937225] F1 signature OK, socitype:0x1 chip:0x4335 rev:0x1 pkg:0x0
[   77.939697] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[   77.940338] CHIP VER = [0x1]
[   77.940338] ----- CHIP bcm4335_B0 -----
[   77.940338] dhdsdio_download_firmware: firmware path=/system/etc/wifi/bcmdhd_sta.bin, nvram path=/system/etc/wifi/nvram_net.txt
[   78.299621] dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.
[   78.439605] dhd_bus_init: enable 0x06, ready 0x06 (waited 0us)
[   78.447113] dhd_wl_ioctl: WLC_GET_VAR: cisdump, ret = -23
[   78.447113] [WIFI_SEC] dhd_check_module_cid: CIS reading failed, ret=-23
[   78.469726] Firmware up: op_mode=0x0015, MAC=xx:xx:xx
[   78.497589] [WIFI_SEC] sec_control_pm: POWER_VAL = 1 
[   78.497619] dhdcdc_set_ioctl: SET PM to 2
[   78.521881] dhd_wl_ioctl: WLC_SET_VAR: aibss_bcn_force_config, ret = -23
[   78.521881] dhd_preinit_ioctls Set aibss_bcn_force_config to 500, 5000, 5000 failed -23
[   78.548522] Firmware version = wl0: Mar  5 2015 16:46:55 version 6.30.171.24.64 (B0 Station/P2P)
[   78.548767] [WIFI_SEC] save .wifiver.info file.
[   78.551208] dhd_wl_ioctl: WLC_SET_VAR: ampdu_hostreorder, ret = -23
[   78.551208] dhd_preinit_ioctls wl ampdu_hostreorder failed -23
[   78.552520] dhd_wl_ioctl: WLC_GET_VAR: pfnlbest, ret = -23
[   78.555938] dhd_wl_ioctl: WLC_GET_VAR: cisdump, ret = -23
[   78.555969] [WIFI_SEC] dhd_check_module_mac: Check module mac by legacy FW : xx:xx:xx
[   78.556152] [WIFI_SEC] Check Mac address in .mac.info 
[   78.562438] ####btlock released, cookie: WiFi
[   78.562438] wl_android_wifi_on() bcm_bt_unlock
[   78.565734] CFG80211-INFO2) wl_cfg80211_attach_post : p2p0: p2p_dev_addr=yy:yy:Yy
[   78.737243] init: Starting service 'p2p_supplicant'...
[   78.744049] smd_pkt_open failed on smd_pkt_dev id:22 - pil_get failed for wcnss
[   79.297912] CFG80211-TRACE) __wl_cfg80211_scan : START SCAN
[   81.853271] CFG80211-TRACE) wl_escan_handler : SCAN COMPLETED: scanned AP count=13
[   82.046051] dhd_wl_ioctl: WLC_IOCTL: cmd: 23, ret = -17
[   82.057250] CFG80211-INFO2) wl_cfg80211_connect : Connectting with zz:zz:zz channel (6) ssid "SSID", len (4)
[   82.057250] 
[   82.096282] ETHER_TYPE_802_1X: ver 1, type 3, replay 1
[   82.096405] wl_bss_connect_done succeeded with zz:zz:zz
[   82.100677] dhdcdc_set_ioctl: SET PM to 0
[   82.115600] ETHER_TYPE_802_1X [TX]: ver 1, type 3, replay 1
[   82.122833] ETHER_TYPE_802_1X: ver 1, type 3, replay 2
[   82.123321] ETHER_TYPE_802_1X [TX]: ver 1, type 3, replay 2
[   82.146392] wl_bss_connect_done succeeded with zz:zz:zz
[   82.157592] CFG80211-TRACE) wl_cfg80211_set_btcoex_dhcp : DHCP is complete 
[   82.233367] CFG80211-TRACE) wl_cfg80211_set_btcoex_dhcp : DHCP session starts
[   82.367950] dhdcdc_set_ioctl: SET PM to 0
[   82.409393] dhdcdc_set_ioctl: SET PM to 2
[   82.434234] CFG80211-TRACE) wl_cfg80211_set_btcoex_dhcp : DHCP is complete 

```

matthiasseemoo, 2017-06-25:
I need remote access to one of your phones, then I can make a patch ready.
If you are interested, just contact me by email.

0xpr03, 2017-06-25:
@matthiasseemoo I'm a student at the TU-Darmstadt, so from what I've seen on your profile I could come by
that should make things easier

matthiasseemoo, 2017-06-25:
Yes, just write me an email to find a date and time.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

harkaz, 2017-07-10:
I'm also very interested in a bcm4335 Galaxy S4 fw patch. As I understand it, bcm4339 and bcm4335 are very similar. Have you started working on it? Is there an ETA for its completion? Will it support injection, as bcm4339 does already, or we should expect monitor mode only? 
Excellent work, keep it up!

matthiasseemoo, 2017-07-11:
so far aron did not contact me so that i could take a look at the firmware.

harkaz, 2017-07-11:
I am keen to help. I will send you e-mail.

matthiasseemoo, 2017-07-11:
i need access to the phone, remotly over ssh to a computer running adb,
connected to the phone. it is sufficient to give me access to a virtual
machine where you connect the phone. however within the next two weeks i
dont have time to work on it.

harkaz, 2017-07-11:
I see. Well, I'll contact you in 3-4 weeks time. I am away from home and I am using my mobile data plan, so using ssh is not easy for me at this moment. But it should be possible after 3-4 weeks.

0xpr03, 2017-07-11:
@matthiasseemoo I've send you an E-Mail.
I'm sorry I've had this in my mind all the time, but some trouble including a laptop crash stopped me from sending anything.

Systemad, 2017-08-05:
The HTC One m7 also uses a bcm4335, if a patch is created for it, can it be applied to all devices with bcm4335 or do you also have to patch it device specifically?
Thanks

matthiasseemoo, 2017-08-06:
As long as the firmware version matches the driver requirements, it is
sufficient to patch one firmware.

chrisdroid, 2017-08-10:
Phones using this chipset (BCM4335):

- LG G2 
- Samsung S4
- Samsung S4 LTE
- HTC One (M7)
- HTC One Mini

chrisdroid, 2017-08-20:
Firmware works well. I think the issue can be closed, and this added to readme.md:
```
WiFi Chip | Firmware Version | Used in                   | Operating System     |  M  | RT  |  I  | FP  | UC  | CT 
--------- | ---------------- | ------------------------- | -------------------- | --- | --- | --- | --- | --- | ---
bcm4330   | 5_90_100_41_sta  | Samsung Galaxy S2         | Cyanogenmod 13.0     |  X  |  X  |     |  X  |  X  |  O 
bcm4335b0 | 6.30.171.1_sta   | Samsung Galaxy S4         | LineageOS 14.1       |  X  |  X  |  X  |     |  X  |  O 
...
```

closed 2017-08-20 by matthiasseemoo

logic11211, 2017-10-17:
how to install on samsung galaxy s4 ???? please...

Ccracker22, 2019-01-05:
Please please please give me a twrp backup rom

dkati, 2019-01-24:
i am the official lineage maintainer of s4 jfltexx. Can i help ?

phmpg, 2021-09-05:
until now, nothing? Has anyone managed to create this patch?

e2002e, 2022-01-15:
The firmware doesn't support injection.