seemoo-lab/nexmon #101: Injection speed?

rodizio1, 2017-07-09:
I have just succesfully compiled and installed nexmon on a Raspberry Pi3 Raspbian 2017-06-21 Kernel 4.9.28.

Injection seems to work, however it's quite slow and unsteady, maximum injection speed is roughly about 300kbytes/s, regardless of datarate used.

Is there any way to get higher injection speeds?

matthiasseemoo, 2017-07-09:
You can always inject at the fastest injection speed by creating frames
directly in the firmware. Unfortunately, our hack in the brcmfmac driver
might not be the best to achieve high injection speeds. If you have some
time, you could try to figure out the bottle neck, which should likely be
somewhere in the driver. And then you could share your results.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

pilotnbr1, 2018-10-13:
Any updates on this?