seemoo-lab/nexmon #102: Error install kali pi0w

4gordi, 2017-07-10:
Hi, i can't install nexmon driver in my pi0w becouse i can't install Linux-headers, I have many errors before install, help me

4gordi, 2017-07-11:
Linux Kali-pi 4.4.50-Re4son+ armv61 Raspberry pi zero w

Re4son, 2017-07-20:
The re4son-kernel comes with headers but you have to install them:

`cd /usr/local/src/re4son-kernel_4*`
`./install.sh -e   ## "e" for hEaders just to confuse everyone :-)`

Best to install the latest "testing" kernel as described here:
[https://whitedome.com.au/re4son/re4son-kernel/](https://whitedome.com.au/re4son/re4son-kernel/)

Please note: kernel builds older than 4.9.24 #20 are cross-compiled and building out of tree modules in Kali might cause problems (I am currently doing a native compile of the 4.4.50 kernel (#20), should be ready tomorrow if you prefer that version)

Please let me know if you encounter any  problems

Re4son, 2017-07-24:
I've updated both the current (4.4.50) and the testing (4.9.24) version of the re4son-kernel for the Raspberry Pi.
Both come with the nexmon drivers fully compiled for armel (Pi0W) and armhf(Pi3). Just install either kernel from here:
[https://whitedome.com.au/re4son/re4son-kernel/](https://whitedome.com.au/re4son/re4son-kernel/)

Reboot, cd into the nexmon sub-directory and follow the steps in nexmon.txt

Happy hacking