seemoo-lab/nexmon #103: Error on RPi3

daggerly, 2017-07-14:
```
COMPILING src/version.c => obj/version.o (details: log/compiler.log)
cat gen/nexmon.pre 2>>log/error.log | gawk '{ if ($3 != "obj/version.o") print; }' > tmp && mv tmp gen/nexmon.pre
ccgcc -fplugin= -fplugin-arg-nexmon-objfile=obj/version.o -fplugin-arg-nexmon-prefile=gen/nexmon.pre -fplugin-arg-nexmon-chipver=`/home/pi/nexmon/buildtools/scripts/getdefine.sh CHIP_VER_BCM43438` -fplugin-arg-nexmon-fwver=`/home/pi/nexmon/buildtools/scripts/getdefine.sh FW_VER_7_45_41_26_r640327` -DNEXMON_CHIP=CHIP_VER_BCM43438 -DNEXMON_FW_VERSION=FW_VER_7_45_41_26_r640327 -DPATCHSTART=$((0x5A2A4 - 0x2000)) -DUCODESIZE=0xB2A3 -DGIT_VERSION=\"2.2.1-69-g2297b-dirty\" -DBUILD_NUMBER=\"$(cat BUILD_NUMBER)\" -Wall -Werror -O2 -nostdlib -nostartfiles -ffreestanding -mthumb -march=armv7-m -ffunction-sections -fdata-sections -I/home/pi/nexmon/patches/include -Iinclude -I/home/pi/nexmon/firmwares/bcm43438/7_45_41_26 -c src/version.c -o obj/version.o >>log/compiler.log
<stdin>:1:64: fatal error: CHIP_VER_BCM43438/patches/include/firmware_version.h: No such file or directory
compilation terminated.
<stdin>:1:72: fatal error: FW_VER_7_45_41_26_r640327/patches/include/firmware_version.h: No such file or directory
compilation terminated.
/bin/bash: ccgcc: command not found
Makefile:55: recipe for target 'obj/version.o' failed
make: *** [obj/version.o] Error 127
```
```
uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
```

lennartkoopmann, 2017-08-07:
I'm getting the same error.

lennartkoopmann, 2017-08-07:
This happens if you upgrade the kernel but don't restart before compiling nexmon. A restart fixed it for me.

closed 2017-08-16 by daggerly

zxy110, 2018-06-02:
hello, I've met the same problem.
This is just because that you haven't configure well. $CC is configured in nexmon/setup_env.sh, so you can take back to   /nexmon , run 
`source setup_env.sh` 
then it can go well.
Remember to be in the same environment.