seemoo-lab/nexmon #105: nexmon: command not found

denverpi3B, 2017-08-07:
Hey, 

I get this error when trying to change modes with nexmon -m2 or nexmon -m0.

What is wrong?

bakashinji, 2017-08-07:
'nexutil -m2' not 'nexmon'

denverpi3B, 2017-08-07:
Sorry. Typo. The error appears when using "nexutil -m2" or "nexutil -m0". I was distracted by the folder name.

bakashinji, 2017-08-07:
see #35 then

denverpi3B, 2017-08-07:
Yes - it fixed it! BUT now… I am not able to connect to access points in managed mode. "nexutil -m0" brings me back to managed mode but I can not connect to access points, as it always says: No wireless interfaces found. 

wpa_supplicant.conf seems to have all the relevant and right data.

bakashinji, 2017-08-09:
ifconfig wlan0 up should fix that you have no interface

Borduni, 2017-08-19:
This issue could be closed

closed 2017-08-20 by DanielAW