seemoo-lab/nexmon #106: Wlan managed, but no internet

djangoary, 2017-08-08:
Everything works fine. Changing from monitor mode to managed works as well. SSH works, but I can not connect to the internet to download and install apps. 

What could be the problem here?

KenHayChung, 2017-08-18:
Don't know what device or OS you are using but I got the same problem with my pi zero W before. In my case I just load the original driver.

rmmod brcmfmac
modprobe brcmfmac

hope this help.

matthiasseemoo, 2017-08-18:
Unfortunately, we start the brcmfmac driver with the wrong interface type. Even though monitor mode is deactivated in the firmware, the interface shows up as radiotap interface. Simply run "nexutil -m0" and you can access regular accesspoints in station mode.

closed 2017-08-18 by matthiasseemoo