seemoo-lab/nexmon #107: Gt-I9060I

mirzaatifbaig, 2017-08-11:
Hi.This phone uses broadcom 4330 according to chroot and some system files.How to enable monitor mode?Sorry Noob Here

matthiasseemoo, 2017-08-11:
simple monitormode would be "su nexutil -m1"

mirzaatifbaig, 2017-11-17:
Sorry I Think It's BCM43430.Any Help?

matthiasseemoo, 2017-11-17:
run "nexutil -V" and tell me the result.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

closed 2018-05-24 by mirzaatifbaig