seemoo-lab/nexmon #110: Is it possible to build a Kali 2017.1 nexmon image for the pi3?

Borduni, 2017-08-15:
Kali 0.1 is outdated and I would try to build a image with the new 2017.1 image. Is that possible?

matthiasseemoo, 2017-08-16:
I think, that the best solution would be to include nexmon directly into Kali, but somehow the Kali developers do not seem to be interested in this approach.

kimocoder, 2017-08-16:
Don't say never, the future will tell if it will be included in Kali NetHunter (",)

Borduni, 2017-08-16:
Let's fork Kali xD

Borduni, 2017-08-16:
Found this https://whitedome.com.au/re4son/re4son-kernel/ - I'll give it a try

kimocoder, 2017-08-16:
There are 2 ways.. the one mentioned above or "BerryBot" which supports Kali, but "sana" is used in /etc/apt/sources.list which needs to be changed to "kali-rolling" and updated for having the latest tools available. Me & some others had some problems with adapters with the r4ason kernel, but with the BerryBot Kali support everything worked fine with both monitor & injection.

So if you experience problems with your way, I can confirm that the BerryBot way works great, but nexmon must be compiled (because it is not included).

BerryBot can be found [here](https://www.google.no/url?sa=t&source=web&rct=j&url=http://www.berryterminal.com/doku.php/berryboot&ved=0ahUKEwiGquHZ9tvVAhUqM5oKHYlNDZ8QFggcMAA&usg=AFQjCNG7DHDQAaGZILWyESwb_AY9gx94Mw)

Borduni, 2017-08-18:
BerryBoot is a installer for Operating Systems, how it may support the monitor mode?

kimocoder, 2017-08-18:
On external adapters there is support, but for internal nexmon must be compiled.

kimocoder, 2017-08-18:
[nexmon bcm-rpi3 here](https://github.com/seemoo-lab/bcm-rpi3)

nethunteros, 2017-08-19:
I have a build script here:

https://github.com/offensive-security/kali-arm-build-scripts/blob/master/rpi3-nexmon.sh

Borduni, 2017-08-19:
@kimocoder I mean the latest 2017.1 version of Kali

kimocoder, 2017-08-19:
Use the build script posted by @nethunteros :)

Borduni, 2017-08-19:
Will that work with the 2017.1 version? Has anyone tested this?

nethunteros, 2017-08-19:
It will download the latest version of Kali Linux. I wrote it/tested it a few months ago.  It should still be working.

devilmind, 2018-07-26:
There a possibility to have the same script for ParrotSec Os ? I'm trying to build a new ParrotSec  image for rpi3 that is including nexmon but no luck!

closed 2020-03-20 by dariotrombello